qmk/qmk_firmware #497: Building on Debian

edasque, 2016-07-08:
I am seeing this odd issue, any idea?
```
~/GMK/keyboards/ergodox_ez# make
-------- begin --------
avr-gcc (GCC) 4.8.1
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: twimaster.c [ERRORS]

/tmp/cc8CkrBq.s: Assembler messages:
/tmp/cc8CkrBq.s: Warning: can't close .build/obj_ergodox_ez_default/twimaster.lst: Input/output error
/tmp/cc8CkrBq.s: Fatal error: can't close .build/obj_ergodox_ez_default/twimaster.o: Input/output error

../../tmk_core/rules.mk:396: recipe for target '.build/obj_ergodox_ez_default/twimaster.o' failed
make[1]: *** [.build/obj_ergodox_ez_default/twimaster.o] Error 1
../../tmk_core/rules.mk:275: recipe for target 'all' failed
make: *** [all] Error 2
```

jackhumbert, 2016-07-08:
Not sure on this - is your drive full?

edasque, 2016-07-08:
No, not really. I am running this on Docker FYI, trying to simplify the flow.

fredizzimo, 2016-07-08:
A quick google search, gave this http://unix.stackexchange.com/questions/39905/input-output-error-when-accessing-a-directory

edasque, 2016-07-08:
It's unlikely my FS is corrupted on Docker. But there is probably an issue with the FS that docker provides here. Curious why it would work in Docker in Vagrant but not "native" docker.

closed 2016-07-08 by edasque