qmk/qmk_firmware #499: GH60 rev.CHN 4th column doesn't work with QMK, but works fine in EasyAVR

stanleylai, 2016-07-09:
Having a weird problem where my 4th column (3, E, D, X) doesn't work when loaded with the default GH60 rev.CHN (aka Satan) QMK firmware, but is totally fine when loaded with EasyAVR instead. Pretty much rules out a hardware problem I think since EasyAVR is fine. Compared the pins in QMK and EasyAVR src and they are both the same.

An interesting issue also is when first plugged in, the keyboard triggers "X" repeatedly until another key is pressed. This goes away when NKRO is turned off. This doesn't happen in EasyAVR.

Got my GH60 from Taobao, and lacks the "intention to do" text but otherwise looks pretty much the same.

Anyone have ideas?

jackhumbert, 2016-07-09:
That would be `C6`, right? That's the audio pin on other keyboards - do you have `AUDIO_ENABLE = yes` in your `Makefile` anywhere? Just to confirm, you're using the `satan` project, right?

stanleylai, 2016-07-09:
Actually the fourth column should be `C7` I think? I have tried disabling all features in the `Makefile` including audio. And yep, using the `satan` project.

jackhumbert, 2016-07-09:
Doy :) either the `C7` port is getting activated somewhere in the code, or there's a short on that port in the hardware. 

If you have RGB and backlight disabled, there shouldn't be any automatic code happening. Are you just doing `make` and not `make quick`?

stanleylai, 2016-07-09:
I wondered if it was hardware related too. But since EasyAVR runs fine on
the same pins, it should be software?

And yep just running make.

sethbc, 2016-07-09:
Did you say you had a split right shift on reddit?  Can you try it using my layout? I'm traveling and don't have access to my satan right now, but mine also didn't have the "intention to do" language (think it just had www.satan.com.cn there) and everything was working fine on it.

sethbc, 2016-07-09:
also i haven't been able to check, but i suppose it's possible 50c686587ed49d8079ba1b11d45ceb6a55d6cd4b may have an impact - mine was working fine when i left.

TerryMathews, 2016-07-09:
https://www.reddit.com/r/MechanicalKeyboards/comments/4rzfo6/help_gh60_revchn_4th_column_doesnt_work_with_qmk/d55geqy

I passed him a known good hex from a fork a0-c did quite a while ago, predated audio support and he had the same problem.

I don't know how to account for why EasyAVR works and QMK doesn't but my gut is telling me this is a hardware fault. Maybe somehow EasyAVR is scanning the matrix differently so that a short isn't causing a problem?

TerryMathews, 2016-07-09:
@sethbc I am certainly not above human error, so I just tested the code from 
https://github.com/jackhumbert/qmk_firmware/commit/50c686587ed49d8079ba1b11d45ceb6a55d6cd4b on my personal Satan and everything is working as expected including the column that @stanleylai is having trouble with.

jackhumbert, 2016-07-09:
@stanleylai just as a sanity check, have you checked the EasyAVR hex since running the QMK one?

sethbc, 2016-07-10:
@TerryMathews thanks very much - i'd have bisected myself but i don't have it with me right now. it didn't look like anything would have impacted functionality, was just thinking back to the last version i knew was good.

stanleylai, 2016-07-10:
I hang my face in shame. After numerous attempts with software fixes, I checked with my multimeter again and turns out that just a touch of solder intended to wire up RGB on the E2 pin shorted C7 as well (which in turn affected the 4th column switches). I took a picture for [posterity](https://lh3.googleusercontent.com/3GLDoo9hbl9LDKIO3UpZ7Wbm02rUhAbC8_nKN2MtWIujahgBeUTBmvGCqtAkbFJuWFwS5yR_rL9lMAZkgnbir8IBvi0DUajNRDJc-ZpPQ7Er2zFZFAIAfmppB7TX8jtkrb1zN9QAG-iE5BUnegNCnlTL5MtIracxvE-7Gjumea5nVl4qqCFpa8wmcFj-N13Izs96EPU9kMVS6Y4AI8S5aodq3Z-81pIyM06Azq2pxnXKsJaFAky9eGWlWLGUJY84gDDb-QhlUw3AqZdqfLIsxyWnBqFOWypzkZaSkjeIWsg6KtXgyLZUtikGNmrK_tXOyhbjVLD2W3mMFsWDFyyoKqGKha9kQEaQgJXADjQ8IZy2jyi7zJ1MiOLIBH6P1IqF1ZKPqKCcCvO5J8e0YvSmhMKDpYM6mkE_ccM3w-ouOgUWDHaZ5Xm0x-YaBCgVRkaX4KwVnij9S4ad2DyC0Op19o9Ivvd5aKx8s_KRv3tYkeh54O3yvbWB8MwkvelsBC3P_x8esCUQ_xc314jvzUZfgtx6KwkmVS0rgpdo38azm7TbGno7AShR-wmjEP6ndHmoe9qccfRrCEHmfeF5JJaTremLgg7H6wUh=w998-h705-no).

@TerryMathews is right though, I am wondering why EasyAVR was fine. QMK didn't work even when RGB was toggled off. Consider this issue closed. Will update the reddit thread with this issue.

TerryMathews, 2016-07-10:
Now that we know the problem, answering that question is simple. Since EasyAVR doesn't support WS2812 or use pin E2 for anything, it's left uninitialized and floating. QMK sets that pin up for digital I/O, so it's no longer floating.

stanleylai, 2016-07-10:
Ah, makes too much sense. Closing the issue out.

closed 2016-07-10 by stanleylai