qmk/qmk_firmware #504: Subprojects are compiled twice

fredizzimo, 2016-07-10:
When running make-all-keyboards, the default subproject is compiled twice. Take a look at [this](https://travis-ci.org/jackhumbert/qmk_firmware/builds/143696695) for example.

You can see that for Clueboard it's doing 

clueboard/clueboard 
clueboard/rev1
clueboard/rev2

And the same thing with Ergodox for example
ergodox/ergodx
ergodox/ez
ergodx/infinity

It's not that bad, since quick compilation is used, so the second pass will be much faster, but still it probably shouldn't do that.

jackhumbert, 2016-07-10:
Yeah, my Makefile skills were exhausted when trying to figure out how to exclude those. The output is also "keyboard/keyboard", which I was trying to avoid, but here we are :)

fredizzimo, 2016-07-10:
Yes, I have had way too much fighting with make files during the past couple of weeks, so I know your feeling. 

I might fix this at some point, but I don't promise anything at the moment.

ezuk, 2016-07-27:
Closing this guy for now.

closed 2016-07-27 by ezuk