qmk/qmk_firmware #508: mapping FN1 (hold) + key (tap)

veshy, 2016-07-11:
id like to use FN1 on the same key as im using Tab. so pressing tab sends tab and holding it down sends FN1

is this possible? im still learning and get the basics done, this is the only thing that im missing for a perfect layout.
thanks in advance

jackhumbert, 2016-07-14:
I don't think this is possible natively - what is your FN1?

veshy, 2016-07-14:
its a momentary layer. i tried ACTION_LAYER_TAP_KEY(3, KC_TAB). im using layer2, the momentary layer is 3. on layer3 the key is set to KC_TRNS
but i cant get it to work

veshy, 2016-07-17:
got it now. used a macro from nicholas keene's ordinary layout and edited it to my needs.

closed 2016-07-17 by veshy

piotr-dobrogost, 2016-07-17:
How does this macro look like? `ACTION_LAYER_TAP_KEY(3, KC_TAB)` should really work here.