qmk/qmk_firmware #509: cant make tap dance work

pvinis, 2016-07-11:
i used it like this https://github.com/pvinis/qmk_firmware/blob/pvinis/keyboards/ergodox_ez/keymaps/pvinis/keymap.c

it should type `A` or `B`, and `C` or `D` with the `BASE` layer top left buttons.

i press them, but they do nothing.

moonrumble, 2016-07-12:
For me it doesn't compile because of unknown field error. I am getting unknown field 'fn' specified in initializer.

pvinis, 2016-07-12:
@moonrumble get the makefile too https://github.com/pvinis/qmk_firmware/blob/pvinis/keyboards/ergodox_ez/keymaps/pvinis/Makefile

moonrumble, 2016-07-12:
Yes, I have those option enable in the Make file. I am not expert on c but it has to do with annonomous struct

pvinis, 2016-07-12:
ok, i figured it out. it was because with the makefile the .hex was in another directory, and teensy still had the old one. i changed the file and it worked. im adding a feature for tap dance. i will make an issue for it, or a pr directly.

closed 2016-07-12 by pvinis

moonrumble, 2016-07-12:
It doesn't compile for me in windows. are you using windows machine to compile the software?

pvinis, 2016-07-12:
im on osx

fredizzimo, 2016-07-12:
@moonrumble,

I don't think it's related to the operation system or compiler version, but rather something in your keymap file, so could you please share it?

moonrumble, 2016-07-15:
I can't compile anything with TapDance feature not only my keymaps. I think it is to do with windows-cygwin and avr compiler.