qmk/qmk_firmware #510: disabling mousekeys disables trackpad on OS X in some circumstances

exiva, 2016-07-11:
I've noticed that when I have my Planck plugged into my MacBook which has the "Ignore built-in trackpad when mouse or wireless trackpad is present" option enabled in Accessibility settings, I cannot use the built in trackpad. Changing MOUSEKEY_ENABLE in the makefile has no effect on it.

jackhumbert, 2016-07-11:
Are you setting `MOUSEKEY_ENABLE = no`? As long as you're doing `make`, that should clean things correctly.

It might help to change the device/product id in the `config.h` - OSX may be holding on to the lufa configuration.

exiva, 2016-07-11:
Just tried changing the device/productid, set `mousekey_enable` to no, `make clean`, `make keyboard=planck dfu` same behavior.

fredizzimo, 2016-07-12:
Which keymap are you using? Are you sure you disable the MOUSEKEY_ENABLE by setting it as 
```
MOUSEKEY_ENABLE = no
```
and not by
```
MOUSEKEY_ENABLE ?= no
```

The latter might get overridden by the default, and the effective value could be set to yes.

You could also confirm that the mouse support really gets disabled by adding the following line
```
#error Mouse support is enabled
``` 
inside one of the MOUSE_ENABLE ifdefs, in the tmk_core/protocol/lufa/descriptor.c file. If the mouse support is properly disabled you should then get no compiler error, but if it's still enabled you will get an error. If no error is generated when compiling you can do a sanity check, by adding this, outside any other ifdef checks
```
#ifndef MOUSE_ENABLE
#error This is a sanity check error, if you get this the mouse support is properly disabled
```

After doing all those tests, remember to revert all your changes, for example by `git checkout tmk_core/protocol/lufa/descriptor.c`

If the mouse support indeed gets disabled, the next thing to check, is that the descriptor.c file in question reports the right thing to the OS.

exiva, 2016-07-12:
I'm writing my own keymap for it. yes, I'm using `MOUSEKEY_ENABLE = no`. I added the sanity check and it built fine. I'll do some more research into looking at what is being reported to the OS, when I have a little more time. I've noticed also the `Ink` system preference also appears, indicating it thinks there's a tablet connected. I will report back more findings as I come across them.

exiva, 2016-07-13:
Okay, I did some more investigating, and edited the lufa descriptor for the mediakeys. on tmk_core/protocol/lufa/descriptor.c#L143 I changed the logical minimum from 0x01 (Pointer) to 0x04 (Joystick) I didn't test moving the minimum higher, but my Planck no longer recognizes as a mouse and a drawing tablet on my Mac, and media keys still work. I can't verify this on Windows however.

Looking at the USB spec it may be possible to change the minimum to 0x80 (System Control) and retain media keys. But I haven't tested this. Joystick seems to work fine and not interfere with my Mac.

Not 100% sure who to submit the patch to (tmk or here to qmk) and it does need more testing.

ezuk, 2016-07-27:
@jackhumbert Do we want the patch here? or should we close and merge from upstream sometime in the future?

jackhumbert, 2017-06-27:
I think #522 addressed this - @exiva, please reopen this if I'm mistaken!

closed 2017-06-27 by jackhumbert