qmk/qmk_firmware #512: A layer switch key for only next key/combo

soelvar, 2016-07-12:
Hi all,
I use intellij a lot and am very dependent on alt/ctrl/shift+FX combos. I tried mapping all fx keys on primary layer, but found myself lacking keys. An awesome feature would be a layer switch key that would only enable the given layer for only the next key/combo.

pvinis, 2016-07-12:
isnt that what `OSL` is for?
from the readme:
OSL(layer) - momentary switch to layer, as a one-shot operation. So if you have a key that's defined as OSL(1), and you tap that key, then only the very next keystroke would come from layer 1. You would drop back to layer zero immediately after that one keystroke. That's handy if you have a layer full of custom shortcuts -- for example, a dedicated key for closing a window. So you tap your one-shot layer mod, then tap that magic 'close window' key, and keep typing like a boss. Layer 1 would remain active as long as you hold that key down, too (so you can use it like a momentary toggle-layer key with extra powers).

soelvar, 2016-07-12:
Thanks for the tip pvinis. I will give it a try tomorrow - hopefully it works with combos like ctrl+fx and not just one key. Are these functions documented anywhere or do I have to find them in the code?

Cheers,
Soelvar

piotr-dobrogost, 2016-07-12:
Implementing OSL for the next combo (as opposed to the next key) would be quite a feat. :)

soelvar, 2016-07-13:
Is there a know bug with the OSL(1), function. When I press the button, it switches to the layer, but never back to layer 0 again?

soelvar, 2016-07-13:
Thanks pvinis for your help. I found the offending line of code:
const uint16_t PROGMEM fn_actions[] = {
  [1] = ACTION_LAYER_TAP_TOGGLE(SYMB)                // FN1 - Momentary Layer 1 (Symbols)
};
It prevented the oneshot from working. I instead created a new layer with just the Fx keys I needed, and the osl with combos like i desired worked perfect:0)

closed 2016-07-13 by soelvar