qmk/qmk_firmware #514: [Infinity Ergodox] - The matrix is sometimes messed up at start-up

fredizzimo, 2016-07-13:
@Lknechtli, reported this in #503, and I'm creating an issue for easier tracking and discussion

> I've noticed that while starting up, the infinity can sometimes get into a weird state where the keybinds don't make any sense. This happens almost 100% of the time when I boot the computer with it connected, but can rarely happen when the computer is already on. I've also noticed a similar problem on the input.club firmware (where the modifiers do not work), but in this firmware it seems like the key matrix isn't being initialized correctly or something. Might be an issue that is more easily exposed during USB driver initialization?
>
> It's not a serious issue, though, since all you need to do is restart the keyboard to fix it.

fredizzimo, 2016-07-13:
I have not noticed this myself, and you are the first one to report it. Quite a few people have run the TMK port before it was ported here to QMK. But there's clearly a problem. So here are some additional questions.

* Is one or both of the sides messed up, or does the problem only affect one of the sides?  
* Can you describe the behavior in more detail, which keymap you are using, and exactly what behavior you see, for example which button you press, the expected output, and what you get?

Lknechtli, 2016-07-13:
Now that I think of it a bit more, the output of the two boards was reversed - right board was acting like it was left and vice versa.
I'm using a keymap that I wrote - 
https://github.com/Lknechtli/qmk_firmware/blob/infinity-colmak-keymap/keyboards/ergodox/keymaps/colemak-dhmod/keymap.c

r2d2rogers, 2016-07-13:
I've seen the reversed (mirrored) layout, but only when I have them plugged in with the wrong master.

Lknechtli, 2016-07-13:
I may have been stupid and done just that - I'll do some testing tonight to see if I can reproduce it again.

fredizzimo, 2016-07-13:
Thanks for the reports. I have a good idea now, of what part of the code, I should read carefully for bugs. 

But we might also deal with some sort of memory corruption, as can be seen in this [issue](https://github.com/fredizzimo/infinity_ergodox/issues/9). There was one that was fixed, but unfortunately the LCD screen still gets corrupted. I have only seen that happen on the QMK port though.

Lknechtli, 2016-07-13:
@fredizzimo in your last comment on that issue:
> It could also be related to the LED support(or the other way around, the LEDs are broken due to this), as according to #5, the LEDs seems to randomly work and randomly not.

Noticed that on my desktop, the left lcd would cut out after a few seconds, but the right one would stay lit
Doesn't happen on my laptop as far as I can tell.

Lknechtli, 2016-07-13:
hm, when connected to my laptop at boot, the left LCD backlight turns off, but if I reconnect it after boot, it stays lit.
:confounded:

fredizzimo, 2016-07-14:
Yes, there are problems with the LCD backlight initialization. But that's unrelated to the LCD screen corruption, and this reported issue. The LCD stuff will be fixed within a few days.

I'm closing this issue for now, if you see problems with the matrix, that is not related to connecting the USB cable the wrong way, please add a comment, and I will re-open it. 

For the corruption, let's continue the discussion in my [TMK fork issue](https://github.com/fredizzimo/infinity_ergodox/issues/9), and for the LED stuff (key LEDs, not the backlight), [here](https://github.com/fredizzimo/infinity_ergodox/issues/5)

closed 2016-07-14 by fredizzimo