qmk/qmk_firmware #517: Issues with the "Remember: These are just aliases" section of the readme

fredizzimo, 2016-07-13:
In the main readme, there's a section called "Remember: These are just aliases"

When reading it, it's very unclear what "these" refers to. I assume that they refer to all the macros inside the "Going beyond the keycodes" section. But in that case it would probably be much clearer if the "These are just aliases" section was moved to the start of "Going beyond the keycodes", and explained in a different way.

I also think that the TMK documentation, which is referred from the "Remember: These are just aliases",  should be merged with the main document, instead of just having a link to the TMK documentation. It is very easy to miss, and the TMK documentation contains a lot of information that I consider required in order to understand the QMK documentation, or at least how to write keymaps.

The other link that links to the TMK documentation is in the "Documentation roadmap" section. The links in that section are referred to as "files you might want to consult", which pretty much states that it's optional reading and as I already wrote above, I consider it mandatory, at least most parts of it.

Note that part of this issue is probably already addressed by #424.

jackhumbert, 2016-07-14:
Yeah, at some point, this section immediately followed the `LT` and friends section and made a lot more sense :)

Agreed on pulling in the TMK docs - I think wording should be changed to make it sound like one cohesive firmware as well.

ezuk, 2016-07-27:
❤️ for opening an issue re the docs! This is awesome. Would you like to submit a PR? Or is there anyone around here who feels like helping with the doc aspect of the project? 

Merging the TMK docs in is quite a task. I am starting to consider migrating over to a better documentation platform, something more maintainable and presentable. This is becoming a book, so merits a book-scale doc framework/tool I think.

fredizzimo, 2016-07-29:
@ezuk Don't you already use Pandoc for the documentation on the external documentation site?

I have just recently started using that at work, and so far I think it's great. You can do pretty much anything by implementing filters. For that I recommend [panflute](http://scorreia.com/software/panflute/). One example of a filter that I made at work, is one that includes other files, and automatically fixes the heading (header in markdown terminology) levels. That was probably less than 50 lines of code.

I will send a pull request that fixes the main issue mentioned above. However generally with the quite limited time that I have, I want to concentrate on the code, especially the Infinity side.

ezuk, 2016-07-30:
For sure -- completly understood. :) And yes, pandoc rocks, definitely an option.

fredizzimo, 2016-07-30:
Closing this now, since the the main issue is fixed and merged

closed 2016-07-30 by fredizzimo