qmk/qmk_firmware #519: Macros and modifiers

fredizzimo, 2016-07-14:
The issue #401 - "Inverting a key", got me thinking about macros and modifiers.

One of the huge drawbacks of the current modifier handling in QMK, is that we are really only able to remove modifiers from macros. Well, it's possible to directly modify the weak, macro, and normal mods, but doing that could easily end up in a mess. In the TMK firmware they added [support](https://github.com/tmk/tmk_keyboard/issues/116) for that. Although that implementation is probably not ideal either, and not directly applicable to the QMK firmware, since macro_mods, have been added, in addition to the weak_ and normal ones.

So, I think first, we need a design, what we want to be able to do from the macros. And remember that we don't necessarily need to support the old MACRO syntax for new features. The advanced macro form is probably much more flexible.

For mods, we probably have several different use cases that need to be taken care of, and different life times of the applied mods. So that requires careful planning.

The implementation would obviously be risky, so therefore I suggest that we first add well covered unit tests to the system(after the unit test framework is [integrated](https://github.com/jackhumbert/qmk_firmware/issues/479), and then develop the required functionality.

I'm also aware that @eltang, had similar thoughts. in this pull reguest #277, and the following issue #272. However I don't believe doing risky changes like that is the way to go, first it needs to be planned properly, and then tested properly.

ezuk, 2016-07-27:
I tend to start thinking about things like that from the interface, syntactic sugar angle. Like, what would a powerful and readable macro call look like? Do you have any thoughts on this?

One syntax I quite like is [AutoHotkey Send command](https://autohotkey.com/docs/commands/Send.htm). I like what they do with the mods there -- using symbols, so ^s is Ctrl+S. I don't know if C lets us do that though.

iFreilicht, 2016-08-15:
> I don't know if C lets us do that though.

Nope, it doesn't, at least not easily. You can in theory write an additional module for the preprocessor (which would have to be separately developed and maintained), but if you wanted to use a symbol that's commonly used in C, you'd have to modify the compiler itself because that's the part the does the lexical analysis. So it's not really worth it. 

The only people I know of doing something like that are Microsoft who implemented the ^ caret, which can be used like a * for pointers, but for special pointers in their environment.

drashna, 2018-10-21:
Is this relevant anymore with `SEND_STRING` and the like?

closed 2019-12-21 by stale[bot]