qmk/qmk_firmware #526: Unable to use 2-setup-environment-win.bat

Whispergod, 2016-07-15:
I'm trying to set up the build environment to create the hex file. I was able to install MHV AVR and MinGW. I ran 1-setup-path-win.bat and I think it succeeded, but when I ran 2-setup-environment-win.bat, a hundred command prompts keep popping up. 

I was a able to set up a build environment on my other Windows 10 laptop without any problems which is pretty weird. Anyone have any ideas on what is going on?

NoahAndrews, 2016-07-18:
I'll be pushing a fix for this soon.

NoahAndrews, 2016-07-18:
Please test the setup scripts in this version of the repository: https://github.com/NoahAndrews/qmk_firmware/tree/fix-script

As of right now it has all of the latest updates from the main QMK branch.

Whispergod, 2016-07-18:
Now I just get two command prompts: One is blank and the other says "Please rerun the script and say yes to the User Account Control popup. Press any key to continue...". I've ran the file with both left click and right click "run as administer" and the result is still the same.

NoahAndrews, 2016-07-18:
Is there anything in `util\log\2-setup-environment-win\elevate.log`?

Whispergod, 2016-07-18:
elevate.log is there, but inside it is blank.
main.log says "'SETX' is not recognized as an internal or external command,
operable program or batch file."

NoahAndrews, 2016-07-19:
Let me know what you get when you download the latest code from [here](https://github.com/NoahAndrews/qmk_firmware/tree/fix-script) again, and run **both** scripts.

Whispergod, 2016-07-19:
Running 1-setup-path-win.bat creates a single prompt that says "Press any key to continue..." 
Running 2-setup-environment-win.bat does the same thing as before: two command prompts and "Please rerun the script and say yes to the User Account Control popup. Press any key to continue...".  The 1-setup-path-win path-setup.log has a text file that is blank. But this time 2-setup-environment-win elevate.log says "Running elevate". Main.log says the same thing as before.

NoahAndrews, 2016-07-19:
Here's some information gathering steps I'd like you to perform.

1. Open a new command prompt (search for it in the start menu / start screen)
2. Give me the output of `echo %PATH%`
3. Give me the output of `setx`
4. The prompt should start out in your home folder (C:\Users\<username>). Navigate to the util folder using the `cd` command. For example, I would type `cd Documents\GitHub\qmk_firmware\util`. Try using the tab key while you're typing that for autocompletion.
5. Once your prompt shows that you're in the util folder, show me the output of `elevate`.
6. Show me the output of `elevate dir`

Let me know if you have any questions about doing all of that.

fredizzimo, 2016-07-19:
I'm quite worried about the use of setx and the other hacks, like restarting explorer of these scripts.

Setx can easily corrupt the path, if it's longer than 1024 characters. [see here for example](http://superuser.com/questions/387619/overcoming-the-1024-character-limit-with-setx). 

I think a much more sensible option would be to make a shell script that is meant to be run from within Mingw, which gets the prerequisites and modifies the path temporarily, until the window is closed. Or alternatively like a bat file, that does the same thing, and then launches mingw, This is the approach taken by for example the Visual Studio command prompt (if you ignore the mingw part).

If the shell script approach is taken, then it could be added to the .profile file, like described [here for example](http://www.mingw.org/wiki/HOWTO_Create_an_MSYS_Build_Environment).  

It could further be combined with the install_dependencies.sh script, with a last action that asks for itself to be added to the .profile, in case it would add some information there to be launched all the time.

NoahAndrews, 2016-07-19:
Thanks for the input!

We actually migrated away from setx after discovering that issue, as discussed in #198. That issue also shows why we need to be restarting Explorer.  At this point, setx is just being used to check for administrator access in the second shell script.

I like your idea a lot. Want to work on it? :D

Whispergod, 2016-07-19:
2. C:\HashiCorp\Vagrant\bin;C:\Program Files (x86)\MHV AVR Tools\bin

3. 'setx' is not recognized as an internal or external command,
operable program or batch file.

5.(c) 2007 - Johannes Passing - http://int3.de/

Execute a process on the command line with elevated rights on Vista

Usage: Elevate [-?|-wait|-k] prog [args]
-?    - Shows this help
-wait - Waits until prog terminates
-k    - Starts the the %COMSPEC% environment variable value and
                executes prog in it (CMD.EXE, 4NT.EXE, etc.)
prog  - The program to execute
args  - Optional command line arguments to prog

6. Opens a blank command prompt and I get a system error "The program can't start because msys-intl-8.dll is missing from your computer. Try reinstalling the program to fix this problem."

NoahAndrews, 2016-07-21:
I unfortunately won't be able to help any further until after this weekend. 

Your response to 6 positively baffles me. That shouldn't involve msys at all.

Whispergod, 2016-07-21:
Take your time. Thanks for the help so far.

ezuk, 2016-07-27:
@NoahAndrews just wondering if you were able to take a quick look at this? :)

kllilizxc, 2016-07-28:
same problem

Primulon, 2016-08-03:
Same issue, outputs are identical to Whispergod.

mmroczka, 2016-08-07:
My issue is similar, and still with the second script, but I get a different related message. "FAILED. Could not find PnPUtil.exe in "C:\windows\System32" or "C:\windows\Sysnative"." I poked around [here](https://m.reddit.com/r/olkb/comments/4pn2eu/problem_with_the_newest_version_of_qmk/) and attempted the "temp folder" solution that they recommended, but no luck the problem still persists. Kinda sucks, cause I was super looking forward to this programming this keyboard. Should I make a separate issue out of this, or is this comment closely related enough to keep the two issues merged?

NoahAndrews, 2016-08-07:
I've been neglecting this thread for waaaay too long. Clearly the scripts need some work, or to be replaced altogether. Unfortunately, I don't really have time to do that right now. 

I think I'll write up some documentation on how to compile the firmware using the brand new Windows Subsystem for Linux (WSL), as a stop-gap measure. That does unfortunately require that you be running Windows 10, with the Anniversary Update that came out earlier this month. 

In the meantime, [here's Microsoft's documentation for setting up WSL.](https://msdn.microsoft.com/en-us/commandline/wsl/install_guide) Once you're in the BASH prompt, the command `cd /mnt/c` will take you to the root of your hard drive. WSL doesn't support USB yet, so you'll need to use [this tool](https://github.com/jackhumbert/qmk_firmware_flasher/releases) to actually flash the firmware. If you  run into issues building the firmware with WSL, make a new issue here (mention @NoahAndrews) or make a new thread on [reddit](reddit.com/r/olkb) (mention /u/SergeantFTC).

If you're not running Windows 10, you can look into the Vagrant or Docker setup instructions, or just run a Linux virtual machine (for now I'll recommend Ubuntu), using VMWare Player or Virtualbox. Hopefully you all find a solution that works for you. Don't be afraid to ask for help.

Please keep in mind that even though I'm basically the only person who has been maintaining and providing support for the scripts since IBNobody rewrote them at the end of March, I am not Jack, nor do I work for him. I do this just because I want to help you guys get your keyboards programmed. :)

Primulon, 2016-08-08:
No problem @NoahAndrews, this is a huge help. 
I'll give this a try once I get Anniversary edition installed.

Primulon, 2016-08-10:
@NoahAndrews - feel free to build/correct off this. 

Pre-requisites:
Windows 10, Anniversary update
Windows Subsystem for Linux enabled: [WSL Install Guide](https://msdn.microsoft.com/en-us/commandline/wsl/install_guide)
dfu-programmer: [dfu-programmer download](https://github.com/jackhumbert/qmk_firmware_flasher/releases)

Run "Bash on Ubuntu on Windows" from start menu 
Command: sudo apt-get update
Command: sudo apt-get install gcc-avr avr-libc dfu-programmer make
Command: cd /mnt/c gets you to the c:/ drive
Follow existing "make" instructions

Drag the .hex file from the desired /keymaps/ folder into dfu-programmer and flash.
Done!

Issues:
If the "flash keyboard" button remains red, check device manager for driver status.
May need to install: [atmel driver](http://www.atmel.com/tools/FLIP.aspx)
In device manager, update ATMega32u4 and manually browse to:
C:\Program Files (x86)\Atmel\Flip 3.4.7\usb

NoahAndrews, 2016-08-11:
Looks good @Primulon.

fredizzimo, 2016-08-20:
I added instructions in #669

zorglups, 2017-01-30:
Hello. I ran into this problem and the hundred of cmd.exe poping up gave me a hard time.
I then opened each BAT, read them and placed the commands manually.
I got it all installed this way.

Unfortunately, the 'make' fails due to the date.exe crashing as reported in several issues.
I then did a "mingw-get install msys" but that did not solve the issue.

I'm having Win7 64bits.

I then went the Docker route.
Not sure I fully understand how to use it though :-(
I did:
1) Launch the "Docker QuickStart Terminal"
2) Waited the end of the update.
3) Ran the following command:
$ docker run -e keymap=default -e keyboard=ergobop --rm -v C/sbin/GitHub/qmk_firmware:/qmk:rw jackhumbert/qmk_firmware

Later, I found that I could run this from a cmd.exe but that it needed the following commands first:
docker-machine env --shell cmd default
@FOR /f "tokens=*" %i IN ('docker-machine env --shell cmd default') DO @%i

----------------------------------
Could you:

1. In https://github.com/qmk/qmk_firmware/wiki#windows-vista-and-later, state clearly that this procedure has problems currently and should not be used (it can loose the PATH variable which can turn into a nightmare).
2. Add more details in the https://github.com/qmk/qmk_firmware/wiki#docker (for newbies like me). I can help, at least on Windows 7 64 bits).
3. Add "mingw-get install msys" to 2-setup-environment-win.bat

Thank a lot for all the job done !!!!

If I can help somehow, please let me know (Should I do pull requests on the Git ???)

NoahAndrews, 2017-01-30:
@zorglups The PATH variable shouldn't be getting corrupted any more, @IBNobody fixed that when he rewrote my scripts. Despite that, I'm in full agreement that we should discourage people from using this method as much as possible.

fredizzimo, 2017-06-03:
For people having problems in Windows 10, there's now a new way of using QMK, through the Windows subsystem for Linux. The instructions can be found [https://docs.qmk.fm/Build-Environment-Setup.html#windows-10].

rimedigit, 2017-06-15:
Hello,

I have the same problem as OP. I run 2-setup-environment-win as administration without success.
When it should download and install dfu-programmer and libusb, it gives me errors. First the "cannot verify <website.url>  certificate," then the "certificate common name <website.url> doesn't match requested host name" errors. 

So I downloaded them both separatedly and tried to install them. I even downloaded and installed dfu-util since the Win-Check-Output log gave me 'not found' for that.

But 2-setup-environment still tried to download and install the dfu and the libusb and errored.

Also, at the end, there is a PnPUtil.exe Not Found message, but I found that that's a misnomer.

I run Windows 7 Ultimate 64 bit. I have installed:

- libusb 1.0.21
- libusb-win32-bin-1.2.6.0 (and I have copied the 4 libusb0 files into the AVRDUDE folder)
- WinAVR
- MHR AVR Tools
- dfu-programmer-win-0.7.2
- I have run "mingw-get install msys" successfully

This is my Win-Check-Output log:
KEY EXECUTABLE LOCATIONS - GENERAL 

Filename = make.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\make.exe
C:\MinGW\msys\1.0\bin\make.exe
Version  = GNU Make 4.0

Filename = git.exe  
Location = C:\Program Files\Git\cmd\git.exe
Version  = git version 2.13.0.windows.1

Filename = cmp.exe  
Location = C:\MinGW\msys\1.0\bin\cmp.exe
Version  = cmp (GNU diffutils) 2.8.7-cvs

-------------------------------------------------------------------------

EXECUTABLE LOCATIONS - AVR MCU COMPILERS 

Filename = avr-gcc.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-gcc.exe
Version  = avr-gcc.exe (GCC) 4.8.2 20131010 (prerelease)

Filename = avr-objcopy.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-objcopy.exe
Version  = GNU objcopy (GNU Binutils) 2.24.51.20131016

Filename = avr-objdump.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-objdump.exe
Version  = GNU objdump (GNU Binutils) 2.24.51.20131016

Filename = avr-size.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-size.exe
Version  = GNU size (GNU Binutils) 2.24.51.20131016

Filename = avr-ar.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-ar.exe
Version  = GNU ar (GNU Binutils) 2.24.51.20131016

Filename = avr-nm.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avr-nm.exe
Version  = GNU nm (GNU Binutils) 2.24.51.20131016

-------------------------------------------------------------------------

EXECUTABLE LOCATIONS - ARM MCU COMPILERS 

Filename = arm-none-eabi-gcc.exe  
Location = Not Found  

Filename = arm-none-eabi-objcopy.exe  
Location = Not Found  

Filename = arm-none-eabi-objdump.exe  
Location = Not Found  

Filename = arm-none-eabi-size.exe  
Location = Not Found  

Filename = arm-none-eabi-ar.exe  
Location = Not Found  

Filename = arm-none-eabi-nm.exe  
Location = Not Found  

-------------------------------------------------------------------------

EXECUTABLE LOCATIONS - NATIVE COMPILERS 

Filename = gcc.exe  
Location = Not Found  

Filename = objcopy.exe  
Location = Not Found  

Filename = objdump.exe  
Location = Not Found  

Filename = size.exe  
Location = Not Found  

Filename = ar.exe  
Location = Not Found  

Filename = nm.exe  
Location = Not Found  

-------------------------------------------------------------------------

EXECUTABLE LOCATIONS - PROGRAMMERS 

Filename = dfu-programmer.exe  
Location = Not Found  

Filename = batchisp.exe  
Location = C:\Program Files (x86)\Atmel\Flip 3.4.7\bin\batchisp.exe
Version  = batchisp version: 1.2.5

Filename = dfu-util.exe  
Location = Not Found  

Filename = teensy_loader_cli.exe  
Location = Not Found  

Filename = hid_bootloader_cli.exe  
Location = Not Found  

Filename = avrdude.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avrdude.exe

-------------------------------------------------------------------------

EXECUTABLE LOCATIONS - OPTIONAL 

Filename = cppcheck.exe  
Location = Not Found  

Filename = doxygen.exe  
Location = Not Found  

Filename = gdb-config.exe  
Location = Not Found  

Filename = wget.exe  
Location = C:\MinGW\msys\1.0\bin\wget.exe

Filename = unzip.exe  
Location = C:\MinGW\msys\1.0\bin\unzip.exe

-------------------------------------------------------------------------

MINGW CHECK - OPTIONAL 

Expected MinGW Base Dir = C:\MinGW

Filename = mingw32-make.exe  
Location = Not Found  

-------------------------------------------------------------------------

MAKE CHECK 

Makfile Found.

-------------------------------------------------------------------------

MAKE CLEAN 

Make Command = make -r -f ../Makefile COLOR=FALSE  atomic-pvc-clean

      0 [main] date 436 handle_exceptions: Exception: STATUS_ACCESS_VIOLATION
  41251 [main] date 436 open_stackdumpfile: Dumping stack trace to date.exe.stackdump
^C

rimedigit, 2017-08-09:
Yes, [Error 5](https://github.com/qmk/qmk_firmware/issues/1466) is my same issue.

drashna, 2018-10-21:
I'm closing this, as this is most likely no longer relevant. 

Between MSYS, and the work on the installation scripts, I don't think this is a problem anymore.

If that's not the case, please reopen.

closed 2018-10-21 by drashna