qmk/qmk_firmware #528: Teensy coma - after flashing, teensy is not recognized (temporarily)

jasonm23, 2016-07-17:
Just posting a note of some strange effects.  I doubt there's enough info here for useful debugging.  I'm posting in case more related flashing issues are reported.

I flashed a Atmega32u4 with QMK the other day and a teensy yesterday.  In both cases after flashing the controller is not seen by the host computer (both cases I was on OSX different machines though.)

The first (not a teensy, a GON Nerd reflashed with a new amtel bootloader.) after the board "bricked" I took it to another host where I have an usbasp and simply reflashed the amtel bootloader onto it. Problem solved.  I can't recall if I tested the board before plugging in the usbasp.  That may have been worth doing, I'll explain that in a moment. Note this board had TMK before attempting to flash QMK, after reloading the dfu bootloader I just flashed my TMK firmware back in place to get an operational board.

Last night I was attempting to flash a teensy based board (hand wired TKL) which already had TMK.  After what looked like a successful flash it seemed the firmware didn't reboot.  I replugged the USB into the same host port and still could not recognize the board.  The light on the teensy was on, but hid listen didn't see a device and attempting to reflash with teensy CLI also reported that there was no device present.

I don't know how to reflash the bootloader onto a teensy so I spent more time googling around for clues and ideas, but nothing was found.

I decided to just try another USB port, and the OS X new keyboard dialog showed up!  hid_listen could see and Shift Shift V showed that QMK was running.

Problem solved!

It seems odd, as I flashed a teensy Phantom a week ago and rebooting was working well. I wonder if there's any reason why this might happen?

 It is very worrying when the board appears to die and replugging on the same USB and doing hard reset (on the teensy button) doesn't remedy the situation.  MacOS USB issues? (El Capitan BTW)

jasonm23, 2016-07-17:
BTW In both cases I was using edited keyboard firmware - reconfigured to match the target boards.

I only changed the matrix pins and keycaps (and Makefile settings.)

jasonm23, 2016-07-17:
I don't mean keycaps (doh)! I meant keymaps! (Autocorrect)

veshy, 2016-07-17:
i am having the same issue with my atreus keyboard, using tmk on a atmega32u4 too. im on windows though.  i just switch to a different host if that happens as using different usb ports doesnt help for me, it just happens once in a while and i cant really reproduce it. never had the issue on a teensy though.

jasonm23, 2016-08-12:
This appears to be a side effect of a counterfeit Teensy 2.0.  See #532

closed 2016-08-12 by jasonm23

jasonm23, 2016-08-17:
New info:  it's basically due to the incorrect bootloader size