qmk/qmk_firmware #531: how do i debug?

pvinis, 2016-07-17:
how can i debug or print debug messages?

i guess i have to enable debug in the makefile, but then how to i connect to it so i see whats printed?

also, can i debug with gdb or something?

skullydazed, 2016-07-17:
You use hid_listen: https://www.pjrc.com/teensy/hid_listen.html

You can't gdb, but you can print messages to be read by hid_listen.

pvinis, 2016-07-17:
awesome. thanks

pvinis, 2016-07-18:
@skullydazed so how do i actually print something?
i need `DEBUG_ENABLE = yes` in the makefile only?

skullydazed, 2016-07-19:
Looks like you figured this out! I'm not sure if your PR meets all the style reqs, but it's definitely better than what's there now. I'll give @jackhumbert and @ezuk a chance to review (or merge) it before accepting.

closed 2016-07-19 by skullydazed

pvinis, 2016-07-19:
indeed I did, and I forgot to close this again. thank you.