qmk/qmk_firmware #532: Teensy / Bootloader/Reset (via Shift + Shift + Pause) doesn't put the teensy in boot mode.

jasonm23, 2016-07-17:
I've found that the doing **Shift + Shift + Pause** isn't putting the teensy 2.0 into bootloader / flashing mode properly. TMK worked fine.

I've looked at #164 and ensured the USB is unplugged/plugged in again to ensure this isn't the cause.

Teensy CLI and the GUI both fail to see it in bootloader mode.

As a side note the keyboard simply begins registering key strokes again in a few seconds after doing **Shift + Shift + Pause**

jasonm23, 2016-07-17:
I reflashed the Teensy after a hardware reset and now the **Shift + Shift + Pause** is working ok.

jasonm23, 2016-07-17:
Setting up a short on the teensy GND / RST that I can trigger easily without opening the case seems like a reasonable long term solution.

jasonm23, 2016-07-18:
@jackhumbert are there options for software based reset?

jasonm23, 2016-07-18:
@jackhumbert one other related question, why does the regular keyboard firmware kick in after 3 or so seconds?  Makes me think it's not _really_ going to the bootloader.

TerryMathews, 2016-07-20:
@jasonm23 What kind of options are you looking for? You can put RESET on your keymap and the sky is basically the limit. I have mine bound to Fn + Backspace.

jasonm23, 2016-07-25:
Hi @TerryMathews I'm basically looking to have keyboard based access to the bootloader.

Shift - Shift - Pause is unreliable, and often necessitates opening up the case. (Sorry if I didn't explain that.)

If mapping RESET on the keymap is possible, and that's identical in operation to pressing the hw button on the teensy,  I think I'll be set.

TerryMathews, 2016-07-25:
Look at the keymap in Phantom for an example of what RESET looks like.

jasonm23, 2016-07-25:
Thanks for the tip!

jasonm23, 2016-07-25:
Place `RESET` on a keymap position.  

The code / pre tags make a big difference in read/cognition (read: certainty) for me.

jasonm23, 2016-07-25:
@TerryMathews - so I mapped up `RESET` on FN0 + Pause which is happening, `hid_listen` borks as expected when it's triggered, but then a few seconds later the keyboard is running the QMK firmware again, and appears to auto-reboot.

This isn't what happens when doing a HW reset and that's what I'm really after.

Is it possible? Am I doing something else wrong? Really at a loss to understand if (a) I have unreasonable expectations of what software reset can/should do. (b) I'm missing some info (c) ???

TerryMathews, 2016-07-25:
You have something wrong in your source then. Reset should be taking you to
the bootloader. If it is not, the bootloader size in Makefile is likely
wrong for the device you're using.

Which, by the way, what device are you working on and can you upload the
source your using to GitHub so we can look at it?

jasonm23, 2016-07-25:
https://github.com/jasonm23/qmk_firmware/tree/ocodo-handwired/keyboards/handwired/tenkeyless

It's a Teensy.

TerryMathews, 2016-07-25:
The Teensy code in Phantom uses 4096 for bootloader size, and I can vouch
that it works correctly on a genuine Teensy 2.0. Try that.

jackhumbert, 2016-07-25:
@TerryMathews really? The bootloader size for the Teensy 2.0 should be `512` :) Maybe it'd work if the bytes before that are all zeros? I'm not entirely sure how the MCU would handle that.

TerryMathews, 2016-07-25:
Phantom drops reliably into Bootloader using 4096. I have to take my
keyboard apart to hit the switch, so making sure RESET worked was a top
priority for me.

I built the Phantom code from Satan, so the bootloader jump carried over as
well and since it worked I didn't alter it. Maybe it's a bit/byte issue
since 4096/8=512?

I can't explain why it works, but it works on both Teensy 2.0s I have.

jasonm23, 2016-07-26:
@TerryMathews @jackhumbert the Teensy is indeed a 512kb memory (I assume that includes the actual bootloader itself.)

Regardless:

```
Teensy Loader, Command Line, Version 2.0
Read "handwired_tenkeyless_default.hex": 23094 bytes, 71.6% usage
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read "handwired_tenkeyless_default.hex": 23094 bytes, 71.6% usage
Programming.....................................................................................................................................................................................
Booting
```

Even though my firmware is only ~23Kb it's reported as 71%.

If I compile the Phantom firmware (let's do that now...)

```
teensy_loader_cli -mmcu=atmega32u4 -w -v phantom_default.hex 
Teensy Loader, Command Line, Version 2.0
Read "phantom_default.hex": 24118 bytes, 74.8% usage
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read "phantom_default.hex": 24118 bytes, 74.8% usage
Programming.............................................................................................................................................................................................
Booting
```
it's also close to 24kb so maybe that clears up the numbers.

jasonm23, 2016-07-26:
After recompiling my firmware both the `RESET` and the <kbd>SHIFT</kbd>-<kbd>SHIFT</kbd>-<kbd>PAUSE</kbd>`  default bootloader trigger are working as expected.  Although that's not the first time.

The difference is that I'm at another location and using a different Mac.  I'll experiment and see if I can make it fail again.

jasonm23, 2016-07-26:
BTW @TerryMathews I'm not sure why your Makefile / bootloader size is working at 4096, mine's unchanged as 512 for the teensy.

No idea why your's is working, because it shouldn't!

jasonm23, 2016-07-26:
Tested with both sizes.  Makes zero difference. SW reset continues to work. (still on the MBPRet)

jasonm23, 2016-07-26:
Ah well, it's not a specific Mac.  I un-plugged the board and reconnected it and SW `RESET` no longer works.

I'm going to try connecting to another host machine now and see how it goes.

jasonm23, 2016-07-26:
Connected to a much older Macbookpro early 2011.  SW Reset not working.

Can remedy only with HW reset. Annoying. :(

TerryMathews, 2016-07-26:
I know why the bootloader size works - learned a ton about ATmega32u4 last night trying to fix a clone Pro Micro.

The bootloader address jump isn't controlled by the software the controller is running, it's controlled by software fuses. Setting the bootloader size bigger than needed shortchanges the availabile program size but ensures maximum comparability.

To your issue, I wonder if your Mac devices that aren't handling soft reset correctly are dropping power to the USB port when the device resets. Do you have a powered USB hub you can use in between the device and the Mac to test?

jasonm23, 2016-07-26:
Hi @TerryMathews after flashing, up until disconnection, the SW reset feature works.  As soon as I disconnect USB or power down the host, SW reset no longer works and a HW reset is required.

No changes on the hardware end affect this. (all Macs; powered/unpowered USB hubs etc.)

TerryMathews, 2016-07-28:
I cannot explain that. That has not been my experience on three Teensy 2.0 based keyboards - 2 Phantoms and a Neutrino.

Do you have a powered USB hub you can try in the middle? I think at this point we've narrowed it down to either the Teensy itself or the behavior of the PCs you have it connected to.

jasonm23, 2016-08-12:
@TerryMathews I have found the answer to this issue, and it turns out to be
a hardware problem.

The Teensy 2.0s that I have been using are apparently clone devices
and not original PJRC Teensies.  I just fitted out a new Phantom TKL
with an official Teensy sourced from mechanicalkeyboards.com.

Notable differences are gold contacts on the pads/through holes and an
orange LED (vs a green LED on the clones)

These were sold on eBay as Teensy 2.0 and come in a solid plastic
packaging, with foam mount and a USB cable included, priced at $10-$15
from a number of vendors, I've also seen them on AliExpress.

They seem to work exactly as a Teensy 2.0 except for this particular
feature.

Hopefully this information will prove useful to someone in the future.

jasonm23, 2016-08-12:
For reference https://www.pjrc.com/teensy/counterfeit.html

TerryMathews, 2016-08-12:
Glad to hear it. Since you've isolated the issue and determined that it's not related to QMK you should close this and the other issue you have open.

I'd be curious to know if there's something fundamentally flawed in the hardware on that clone or if it's just a bad implementation of the Teensy bootloader. Would you be willing to pass it on to me for shipping cost? I've got the hardware to flash a different bootloader on via ISP and see if it behaves any better.

jasonm23, 2016-08-12:
I have a few of them, so I'd be happy to send one to you.  ping me via my username at Google's popular email system.

closed 2016-08-12 by jasonm23

jasonm23, 2016-08-17:
BTW verified - Teensy genuine the bootloader size set to 2048 works.

512, 1024, 4096 do not work. All tested.

AMtel bootloader and the LUFA bootloader, 4096z

Yoobe, 2016-10-09:
Hi @jasonm23 ,

I built my first neutrino this week end and I have the same issue for reset of the teensy.. I came here to understand that I dont have a copy of teensy 2.0 :(
Did you find a solution for the fake teensy reset? 
I cannot find a genuine version of teensy 2.0 anymore. All sellers are out of stock.

TerryMathews, 2016-10-09:
You can make the clone Teensy work reliably if you reflash the bootloader.
I have one that should work I can send you, but you need an ISP programmer.

Yoobe, 2016-10-09:
I have a JLINK  and a ST-link.. don't think they are compatible with AVR boards :(
Any usb/ttl board is not useful? I have some FTDI boards...

Yoobe, 2016-10-09:
Maybe I could use an arduino board as ISP

I have a Arduino Nano V3.0 and based on this website (https://www.arduino.cc/en/Tutorial/ArduinoISP) I can try to flash the boatloader..

Yes I am interested by the bootloader you have

jasonm23, 2016-10-09:
Yoobe the amtel bootloader is fine, you can get it from their site.

http://www.atmel.com/devices/atmega32u4.aspx?tab=documents

Sent from my iPhone

TerryMathews, 2016-10-10:
Yoobe, what's your e-mail? I can send you what I sent to JasonM23 that he
was able to successfully flash and use his cloned Teensy normally. I should
point out that you won't be using the Teensy tools anymore, this is an
Atmel bootloader and as such you'd be using DFU to flash going forward.

jasonm23, 2016-10-10:
Terry did you modify it?  Yoobe should just download it from amtel.

jasonm23, 2016-10-10:
Btw, thanks for letting me know you got the teensy clone in the mail.

TerryMathews, 2016-10-10:
It's not from Atmel, it's from http://www.stanleylio.com/home/usblufa

And yes it is modified, because as-is it wasn't reliably soft resetting
into Bootloader. I had to flip the logic on the default state based on the
way my Pro Micro clones were wired. It seemed to work for you as well.

Finally, no I didn't receive the clone you mailed me. I didn't think it was
worth mentioning.

Yoobe, 2016-10-10:
here is my email: yoobe@ersezer.com
Thanks for the bootloader.

stefanadelbert, 2017-04-19:
I am having this same problem with a "Teensy 2.0" bought from AliExpress. If I perform a hardware reset then I am able to perform software resets until I unplug. On replugging, software resets no longer jump successfully to the bootloader - it seems that the firmware is restarted. I was getting similar behaviour with TMK. I have tried a few different values for BOOTLOADER_SIZE (4096, 1024, 512).

jasonm23, 2017-05-02:
@TerryMathews Sorry for the late reply.  It was sent tracked and according to the carrier it was delivered / received.

jasonm23, 2017-05-02:
@stefanadelbert as you'll read above these Teensy's from AliExpress are not the real deal, so you need to reprogram them with a USB-ISP programmer. 

They look something like this.

[image]

Just google for 'usb isp programmer' and you will get a bunch of options to buy one.  They are cheap, often less than $10.

You can then re-flash the bootloader as per Terry's instructions and my notes above.

stefanadelbert, 2017-05-02:
@TerryMathews Thanks for the reply - I really appreciate it. I ended up drilling a hole in the bottom of my keyboard so that I could access the reset pushbutton on the "Teensy". It's shame, but it works. Next time around I will get myself the real thing instead of the knockoff.

jasonm23, 2017-08-19:
NOTE: using a BOOTSIZE of 2048 with these teensy clones (instead of 4096) allows soft reset using LShift + RShift + Pause