qmk/qmk_firmware #544: Missing keymap_midi.h header

Polymetric, 2016-07-22:
Getting an error when compiling current master with the default Planck layout.

Steps to reproduce:
Checkout/clone current master.
Edit Makefile in `keyboards/planck/default` to include `MIDI_ENABLE=yes`
`make`

I get an error indicating that the header file is missing. Looking at Google cache, it appears the header was present at some point, and still is on the rev5 branch, although the rev5 branch also fails to compile.

jackhumbert, 2016-07-25:
Sorry about that! Fixed in 4aa7d2e - would you mind pulling the changes and trying them out?

anhdat, 2016-07-25:
@jackhumbert did you mean [this commit](https://github.com/jackhumbert/qmk_firmware/commit/8b94e26d7c3b30cc57d710a11e5651d15e8e3b20). Still I get warning when compile:
```
Compiling: ../../../../tmk_core/protocol/lufa/lufa.c                                                [WARNINGS]
 |
 | ../../../../tmk_core/protocol/lufa/lufa.c:89:5: warning: excess elements in struct initializer
 |      usb_send_func,
 |      ^
 | ../../../../tmk_core/protocol/lufa/lufa.c:89:5: warning: (near initialization for 'lufa_driver')
 | ../../../../tmk_core/protocol/lufa/lufa.c:90:5: warning: excess elements in struct initializer
 |      usb_get_midi,
 |      ^
 | ../../../../tmk_core/protocol/lufa/lufa.c:90:5: warning: (near initialization for 'lufa_driver')
 | ../../../../tmk_core/protocol/lufa/lufa.c:93:1: warning: excess elements in struct initializer
 |  };
 |  ^
 | ../../../../tmk_core/protocol/lufa/lufa.c:93:1: warning: (near initialization for 'lufa_driver')
 | ../../../../tmk_core/protocol/lufa/lufa.c: In function 'midi_usb_init':
 | ../../../../tmk_core/protocol/lufa/lufa.c:757:3: warning: implicit declaration of function 'SetupHardware' [-Wimplicit-function-declaration]
 |    SetupHardware();
 |    ^
 |
```

and error when deploy:

```
Bootloader Version: 0x00 (0)
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Bootloader and code overlap.
Use --suppress-bootloader-mem to ignore
make: *** [dfu] Error 4
```

jackhumbert, 2016-07-25:
Warning should be ok - I'm not sure why that's there though. I get the same on my machine.

It looks like your .hex is too big - you'll need to turn off some of the features and recompile in order to use it :/

anhdat, 2016-07-25:
Yeah, disabling backlight feature makes this work. 👍

Polymetric, 2016-07-25:
Totally working now. Thanks!

closed 2016-07-25 by Polymetric