qmk/qmk_firmware #545: SEQ_TWO_KEYS is not working but SEQ_ONE_KEY is working as expected.

moonrumble, 2016-07-22:
I want to use leader functionality. I am following all the steps explained in the read me file. It works for one sequence but it doesn't work for 2 sequence.

I tested with another keymap (@algernon) it works fine. It seems like there must be something wrong in my keymap. Here is my [KeyMap](https://github.com/moonrumble/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/apatel/keymap.c)

I can't figure out what am i doing incorrectly?

Thanks

hia3, 2016-07-24:
You have `MEH_T(KC_A)` in your `BASE` layer, but you use just `KC_A` in `SEQ_TWO_KEYS` macro.

Try using `SEQ_TWO_KEYS(MEH_T(KC_A), KC_B)` or change `MEH_T(KC_A)` to `KC_A` in `BASE` layer.

moonrumble, 2016-07-25:
No, This is not how it works. I already tested with `SEQ_ONE_KEYS` and `S` which works S is mapped to `SFT_T(KC_S)`

moonrumble, 2016-07-25:
I don't think it is my keymap file. I have just tried following

`SEQ_TWO_KEYS(KC_A, KC_B) {
        			SEND_STRING("a  b");
        }`
`SEQ_TWO_KEYS(KC_P, KC_O) {
        			SEND_STRING("a  b");
        }`

Leader -> PO works but Leader-> AB doesn't work. is it because of modifiers associated with A and B? 

@jackhumbert  @ezuk @eltang is it possible for any experts to look at my keymaps to shade some light?

algernon, 2016-07-25:
The `SEQ_TWO_KEYS()` macros check the keycodes, the ones that appear in your keymap. If you have `MEH_T(KC_A)` in the keymap, that is **not** the same as `KC_A`. I'm not sure how the `_T` functions work together with leader, it may be they do not work well together at all.

The issue here is likely not with `SEQ_TWO_KEYS`, but with the leader in combination with the `_T` stuff. I'm using `SEQ_TWO_KEYS`, and it works fine. Looking at the code, there is no reason it should not work when `SEQ_ONE_KEY` works.

Can you try `SEQ_ONE_KEY(KC_A)`? I'm pretty sure it won't work, either.

moonrumble, 2016-07-25:
`SEQ_ONE_KEY(KC_A)` doesn't work either. so it seems like I can't use Leader functionality with any of my home row keys. 

`SEQ_TWO_KEYS(MEH_T(KC_A), SFT_T(KC_S)) {
            SEND_STRING("abc");
        }
`
Above is working.  @hia3 Thanks , your suggestion worked.

algernon, 2016-07-25:
The thing to remember about Leader, is that it is based off of the keycodes in your keymap, and not off the codes they end up sending.

moonrumble, 2016-07-25:
It is really puzzling how `MEH_T(KC_A)` is even working because I am pressing Leader->KC_A

closed 2016-07-26 by moonrumble