qmk/qmk_firmware #547: KC_LGUI and KC_LALT switched

emombay, 2016-07-24:
Hey everyone. 

I was getting this issue of my Ergodox EZ, but was hesitant to report this as I didn't have other hardware to test. Using the latest codebase from master, compiling for both Ergodox and Preonic, I would get switched output for LGUI and LALT. Basically pressing a key bound to LGUI would behave as if I pressed LALT, and vice versa. I'm using a Windows 10 machine on a KVM, and I get the same symptoms on my work laptop with Windows 7.
My workaround has been to create a macro that presses LGUI and LALT and binding the keys to these macros. Any ideas what's going on?

djiess, 2016-07-24:
Tested with a preonic at 8e1d96983aad19b3e4e72ece5db822e92d91342b. On linux, xev tells me that those 2 keys work as expected.

jackhumbert, 2016-07-25:
If you have bootmagic enabled, could you try pressing both space and backspace at the same time when plugging it? It sounds like the switch-alt-and-gui setting is enabled!

emombay, 2016-07-25:
I copied over the default Makefile which had bootmagic disabled. Any way to switch this setting without using bootmagic?

jackhumbert, 2016-07-25:
You can use the keycode `AG_NORM` (like [here](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/planck/keymaps/default/keymap.c#L164)) if you can reflash your keyboard!

emombay, 2016-07-25:
I see. Wonder why this is also happening in my ErgoDox EZ, which I believe had it off. Didn't really mess with bootmagic on that, but somehow the same symptoms occurred.

jackhumbert, 2016-07-25:
BM is often triggered accidentally - I think the way to swap them is space + alt when plugging it in :/

emombay, 2016-07-26:
Finally had a chance to test this. For Preonic, I had to press AG_SWAP to get the two keys to behave normally. Shouldn't this come by default? I've customized my layout so I couldn't get space + alt to work.

emombay, 2016-07-27:
@ezuk I've flashed it by using my macro workaround. Will try and flash it back to standard later this evening, and try the bootmagic fixes. I'll update this right afterwards.

emombay, 2016-07-28:
@ezuk I reflashed my EZ to use LGUI and LALT properly. However, I couldn't get the bootmagic settings to work, and was not able to reset their behavior (I tried alt + space during bootup, and key behavior didn't change). I did make it work by using the `AG_NORM` and `AG_SWAP` keycode and setting that in a key in another layer. Pressing `AG_NORM` set the key behaviors back to normal. 
I can probably mark this closed, as I am ok now, unless otherwise.

ezuk, 2016-07-29:
Thanks @emombay! Closing this then.

closed 2016-07-29 by ezuk