qmk/qmk_firmware #560: Windows, Ergodox, KC_SLEP

hia3, 2016-07-26:
Looks like after [this](https://github.com/jackhumbert/qmk_firmware/commit/d48a4cf1eb80600fb3e8eb859e5b77c954a18137) commit KC_SLEP stopped working on Windows.

ezuk, 2016-07-27:
Hi @hia3! Would you mind providing some more detail? Which version of Windows, maybe your keymap or repo? Or if you have positive proof that this is the issue, a PR with a possible fix?

hia3, 2016-07-27:
Sure. I'm building with Ubuntu (VirtualBox):

mkdir repro && cd repro/
git clone https://github.com/jackhumbert/qmk_firmware.git
cd qmk_firmware/keyboards/ergodox_ez/
make keymap=ordinary

And test on Windows 10: press and hold "[Media Tab](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/ordinary/keymap.c#L46)" button (3rd led will turn on). Then press [Sleep](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/ordinary/keymap.c#L132) button - nothing will happen.

Changing all 0x0081 back to 0x0001 in tmk_core/protocol/lufa/descriptor.c helps.

Btw, regardless of this commit I've never been able to wake up the PC with ErgoDox EZ (even though waking up with another keyboard works just fine).

ezuk, 2016-07-29:
@jackhumbert what do you think? Seems like a good isolation of a bug imho. Also, can people wake up Windows machines with the Planck? Trying to understand if this is keyboard specific.

jackhumbert, 2017-06-27:
It looks like #963 addressed this - @hia3, if this isn't the case, please reopen this!

closed 2017-06-27 by jackhumbert