qmk/qmk_firmware #574: tap-dance: holding a key triggers on_finished every TAPPING_TERM

algernon, 2016-07-27:
When a tap-dance key is held, the `on_finished` callback is triggered from the matrix scan every TAPPING_TERM. Ideally, it should only trigger once.

The easiest would be to have a bool that stores whether it was triggered before, but perhaps there's a better way. And in any case, this should only be worked on, once the other tap-dance issues are ironed out, as to not make things worse. Recording this here as an issue, so I don't forget to deal with this too at some point.

cdlm, 2016-07-27:
BTW, `process_tap_dance.h` needs to `#include "action_tapping.h"` else `TAPPING_TERM` isn't defined

algernon, 2016-07-27:
Thanks, will add that too!

PR #645 (2016-08-18): tap-dance rework
This reworks how the tap-dance feature works: instead of one global state, we have a state for each tap-dance key, so we can cancel them when another tap-dance key is in flight. This fixes #527.

Since we have a state for each key, we can avoid situation where a keyup would mess with our global state. This fixes #563.

And while here, we also make sure to fire events only once, and this fixes #574.

There are two breaking changes, though: tap-dance debugging support was removed, because dumping the whole state would increase the firmware size too much. Any keymap that made use [... cut]
commit 29f64d7a (2016-08-18): tap-dance: Major rework, to make it more reliable

This reworks how the tap-dance feature works: instead of one global
state, we have a state for each tap-dance key, so we can cancel them
when another tap-dance key is in flight. This fixes #527.

Since we have a state for each key, we can avoid situation where a keyup
would mess with our global state. This fixes #563.

And while here, we also make sure to fire events only once, and this
fixes #574.

There is one breaking change, though: tap-dance debugging support was
removed, because dumping the whole state would increase the firmware
size to [... cut]
closed 2016-08-18 by jackhumbert