qmk/qmk_firmware #581: MEH_T(KC_PLUS)) does not produce +

nrrkeene, 2016-07-29:
The KC_PLUS alias for LSFT(KC_EQL) works fine for me but MEH_T(KC_PLUS) produces an equal sign. I have to press my own shift key to get the plus which isn't what I want to happen. Similarly, the other aliases seem to give the same, such as 7 instead of ampersand for MEH_T(KC_AMPR).

ezuk, 2016-07-29:
Just to confirm: On a regular keyboard, what happens when you press Alt+Shift+Ctrl+7?

algernon, 2016-07-29:
I just tried it on a TypeMatrix, and on the laptop keyboard, and it did not produce any input at all. (I'm on Linux, mind you. Tried it in the browser and in an xterm too)

nrrkeene, 2016-07-29:
To be clear, what I expect is to get PLUS when I tap the key and MEH if I hold it down for a key combo -- just like CTL_T is used on the default layout.

When I press Alt+Shift+Ctrl+7 I get nothing (on Linux and Mac) because that key combo does not trigger an action. I wouldn't expect a char to be printed, and it isn't.

I looked at the file which creates the KC_PLUS alias and I don't really understand all the bitshifting that is done. I suspect that the LSFT() is lost, leaving MEH_T(KC_PLUS) to have the same meaning as MEH_T(KC_EQL). If that is simply fundamental and can't be overcome then I will not use MEH_T in that context, I'll just use straight KC_PLUS, which is almost as good for me.

jackhumbert, 2016-08-08:
KC_PLUS is a 16bit keycode, and MEH_T and friends only work with 8bit keycodes right now! This is definitely something we need to address in the documentation :)

closed 2016-08-08 by jackhumbert