qmk/qmk_firmware #585: pop layer

pvinis, 2016-07-29:
so i was thinking i might need this:
- press a key, and `TG(SYSCTL)` happens.
- on `SYSCTL` keymap, i want to have a few keys that do something, like volume up or something.
- all other keys, i want them to pop the `SYSCTL` layer.

is there a way to do that now, except for putting all other keys in that layer to `TG(SYSCTL)`?

i thought i could add a new `KC_POP`, that would pop the layer you are on. would that make sense?

ezuk, 2016-07-29:
this sounds like one-shot layer keys. Check out [this part of the readme](https://github.com/jackhumbert/qmk_firmware#switching-and-toggling-layers) - `OSL` specifically. Sounds like what you're looking to do?

pvinis, 2016-07-29:
no.. when im in the `SYSCTL` layer, i want to be able to press vol up and down as many times as i want. and only when im done, i can press the same `toggle` button i used to go in that layer, to go out of it.
but i would like to press any other key to go out of it.

pvinis, 2016-07-29:
now that i think about it, we could even have a `KC_POP` that pops the layer, and a `KC_POPANDSEND` that pops the layer, and also does whatever the key in the layer below would do.
so in my example, i go to layer `SYSCTL`, put the volume to max, then if i have `KC_POP` on a the `Y` key, for example, i press `Y` and go to `BASE` again. then i can type `YEA!`. in this case, i tap on `Y` twice.
if the `Y` button on the `SYSCTL` layer is `KC_POPANDSEND`, then after i max out the volume, i tap `Y` and i immediately fall back to `BASE`, and also a `Y` appears on the screen, so i can keep writing by just tapping `EA`.

how does that sound?

algernon, 2016-07-29:
Untried idea: use `OSL`, but make the volume up/down/etc keys macros, which reset the one-shot state, and re-install `SYSCTL` as a one-shot layer. Pressing any other key pops the layer, but pressing the layer keys, the layer is re-instated, and remains in effect.

Other than this, some `process_record_user` things may do what you want... or perhaps you could use that, instead of macros.

Mind you, I haven't tried, no idea if this'd even work.

pvinis, 2016-07-29:
hm, i will try like that and report back. but it feels a bit more weird.. especially if i have more than a few buttons (which i will).

i might try the way i proposed and see how it works etc.

iFreilicht, 2016-08-12:
How were you results? Could you implement everything inside the keymap?

pvinis, 2016-09-21:
havent finished it yet. i need to do it in a weekend. i will update when its done.

closed 2017-06-27 by skullydazed