qmk/qmk_firmware #598: No device present. Bootloader not found

denolfe, 2016-07-31:
I am getting this error on both my Windows and Ubuntu-based machine when attempting a `make dfu`:

```
avr-gcc (GCC) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  17292	      0	  17292	   438c	satan_default.hex

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  17292	      0	  17292	   438c	satan_default.hex

dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
```

I am only able to find dfu-programmer 0.6.1 in my linux repos. I attempted to compile 0.7.2 to see if that was the issue, but I was unsuccessful in doing so.

Any help is appreciated, thanks!

exiva, 2016-08-03:
in `tmk_core/avr.mk` try commenting out lines 110-113. I only tested #521 on a Planck, but perhaps whatever mcu satan uses doesn't respond to a bootloader version request.

denolfe, 2016-08-03:
@exiva Thanks for the idea, but it had no effect on whether the device was found.

denolfe, 2016-08-03:
This has to be a driver issue as my firmware compiles fine. I also can see my board when I put it into DFU mode. `lsusb` outputs `Bus 001 Device 046: ID 03eb:2ff4 Atmel Corp. atmega32u4 DFU bootloader
`. Maybe I'll take a look at avrdude or easyavr to see if I have better luck.

denolfe, 2016-08-03:
So I got my keyboard to flash by using Git Bash on Windows. However, the device does not show up for me in the AVR Shell or in Ubuntu.

[image]

kuel, 2016-08-07:
This confused me for a while too, try to run it in sudo, so "sudo make dfu"

denolfe, 2016-08-08:
@kuel Thanks, I'll give that a try tonight. If that's the case, it would be a bit of an oversight by me :sweat_smile:

denolfe, 2016-08-08:
@kuel Unfortunately, running it in sudo did not make a difference :cry:

kuel, 2016-08-08:
Just one last idea from me, are you pushing the button on the keyboard, the one to put it in dfu mode (it's on the back)?

kuel, 2016-08-08:
Does it work in just regular CMD? Other USB ports maybe (sometimes USB 2 or 3 can cause problems with certain devices)?

denolfe, 2016-08-08:
I've tried letting `make dfu` put the device into dfu mode as well as pushing the button. Both yield the same result.

Interestingly, the dfu-programmer erase command was able to blow away my config. I'm pretty confused at this point.

kuel, 2016-08-08:
Try flashing it with the default .hex ps://github.com/jackhumbert/qmk_firmware/releases/download/v4.0/planck_pcb_r4.hex

then run
dfu-programmer atmega32u4 erase --force
dfu-programmer atmega32u4 flash planck_pcb_r4.hex
dfu-programmer atmega32u4 reset

thaddeusreid, 2016-08-09:
I landed here after experiencing a similar issue flashing my Planck on my Ubuntu 16.04 machine. 

Cloned the QMK repo, navigated to my respective keyboard (` /keyboards/planck/` ), and successfully ran `make`. Hitting the Reset button made the keyboard properly appear in `lsusb`. When I attempted to run `make dfu` the process suspended with this message:
```
dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
```

I read through this thread and tried a few of the recommendations. `sudo make dfu` had the same results as above. Next I tried `dfu-programmer atmega32u4 erase` but this too had an error: `dfu-programmer: no device present.`

However adding a `sudo` to the commands @kuel listed did the trick:
```bash
sudo dfu-programmer atmega32u4 erase
sudo dfu-programmer atmega32u4 flash planck_pcb_r4.hex
sudo dfu-programmer atmega32u4 reset
```

Happily typing away on my new keymap!

exiva, 2016-08-10:
If Linux requires sudo to use `dfu-programmer`, try this tmk_core/avr.mk which should prompt for sudo access on `make dfu`and use it. (note: I haven't tested this out on Linux, and I just bodged it together.) https://gist.github.com/exiva/0c353e2a940eefcf3a6dfeb897861333

denolfe, 2016-08-11:
Well, I was confused when I said I was able to do the `erase` command on my board. I was able to do it on *Windows*, but my board is still unrecognized in Linux regardless of `sudo`. Thanks for all of those that have chimed in this far, though.

denolfe, 2016-08-11:
Well, after all the struggle, pressing the hardware button on the back does indeed put it into the correct state for flashing. Not sure why `make dfu` cannot put it into that state on its own, but not a big deal at all. Thanks all again.

closed 2016-08-11 by denolfe

drudru, 2017-12-28:
@thaddeusreid - your answer worked for me

I got a kbdfans assembled GH60 keyboard (which I'm typing on right now!)

make with the dfu would just fail.

After doing your erase, flash, reset... everything worked great. 

Thanks!

chamster22, 2018-04-29:
[image]
I've been using Msys2 do you know how to solve this?

ecumene, 2020-03-14:
I've run into this while following the more recent qmk command line instructions for linux. If you're using `qmk flash`, follow these steps to setup qmk using root since `dfu-programmer` requires superuser:

1. Install `qmk` and ensure `sudo qmk` works
2. Run `sudo qmk setup` (this installs the qmk firmware in `/root/qmk_firmware`)
3. Copy your firmware: `sudo cp ~/.../asdf_asdf_default.bin.hex /root/qmk_firmware`
4. Flash the firmware: `sudo qmk flash -kb asdf/asdf -km default`

Worked for me first try!