qmk/qmk_firmware #601: LT(layer, KC_LGUI) not working as expected

Asetnioh, 2016-08-01:
I'm currently customizing my Ergodox EZ. I was previously using the rather limited massdrop configurator.

I'm using a key on the base layer that is supposed to send me to a layer if held, and send KC_LGUI if tapped. It doesn't seem to send the tap key at all, and the keys in the layer don't work as expected either. Have I done something incorrect here?

I've attached the keymap. It's extremely untidy.
[keymap.txt](https://github.com/jackhumbert/qmk_firmware/files/393310/keymap.txt)

common-nighthawk, 2016-08-16:
I'm not positive, but this may be because both GUIL and KC_LGUI are modifier keys.  It makes sense to me that a modifier key is not elibible to be 'tapped'.  Perhaps try `LT(GUIL, KC_1)` to see if that's the problem.

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @Asetnioh please reopen this or open a new issue.

closed 2017-06-27 by skullydazed

salas0316, 2017-09-14:
I have same issue here (AMJ60)
LT(layer, KC_LCTRL) -> wont send any key
LT(layer, KC_LSHIFT) -> send Ctrl
LT(layer, KC_LALT) -> send Shift
LT(layer, KC_LGUI) -> send Shift
LT(layer, KC_RCTRL) -> send Alt
...
I can set KC_LSHIFT to get Ctrl key pressed, but have no luck to get gui key in anyway.
All I can do is bind all key to LGUI(kc). it's ugly.
And for MO() TG() ... it goes to another layer, but can't go back to default layer.