qmk/qmk_firmware #602: Keyboard not working: infinity_chibios

F1LT3R, 2016-08-02:
I am unable to build the `infinity_chibios` keyboard using either:
- The `README.md` [make instructions](https://github.com/jackhumbert/qmk_firmware#the-make-command)
- The [infinity_chibios/MEMO.txt file](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/infinity_chibios/MEMO.txt)

Here is my output:

```shell
~/repos/qmk_firmware/keyboards/infinity_chibios   master  make
../../tmk_core/chibios.mk:38: ../../lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x5.mk: No such file or directory
../../tmk_core/chibios.mk:40: ../../lib/chibios/os/hal/hal.mk: No such file or directory
../../tmk_core/chibios.mk:46: ../../lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: No such file or directory
../../tmk_core/chibios.mk:56: ../../lib/chibios-contrib/os/hal/boards/MCHCK_K20/board.mk: No such file or directory
../../tmk_core/chibios.mk:57: ../../lib/chibios/os/hal/osal/rt/osal.mk: No such file or directory
../../tmk_core/chibios.mk:59: ../../lib/chibios/os/rt/rt.mk: No such file or directory
../../tmk_core/chibios.mk:65: ../../lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: No such file or directory
../../tmk_core/chibios.mk:67: ../../lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
make: *** No rule to make target `../../lib/chibios/os/hal/lib/streams/streams.mk'.  Stop.
```

Any advice for moving beyond this point?

Depariel, 2016-08-02:
If you navigate to `qmk_firmware/lib`, you ought to see three folders there: `chibios`, `chibios-contrib`, and `ugfx`. They will likely be empty, which is why it's saying that those files can't be found (the `No such file or directory` error).

You need to make sure that when you `git clone` the repo, you also get those submodules in `qmk_firmware/lib`. When cloning the whole repo in the first place, you should be able to use `git clone --recursive <repo>`. If you've already cloned it, then `git submodule update --init --recursive` once you've `cd`'d into your `qmk_firmware` directory should get those submodules for you. Otherwise you can just download them manually and copy the files into the appropriate folders.

This should probably be added specifically in the documentation somewhere to be extra clear that those submodules in `qmk_firmware/lib` aren't hosted directly in the qmk_firmware repo.

F1LT3R, 2016-08-02:
Ok that's great, thanks @Depariel. It makes the hex file now.

finleye, 2016-08-08:
@Depariel is the infinity_chibios intended for the Infinity 60% from Massdrop?

F1LT3R, 2016-08-08:
@finleye I don't think so. I couldn't flash my Mass Drop Ergodox from Input Club with that build it with that build. So far I've had the most success with the [kiibohd controller repo](https://github.com/kiibohd/controller). Specifically, the `keyboards/ergodox.bash` repo builds and flashes correctly for me.

finleye, 2016-08-08:
@F1LT3R thanks for the info. I'm looking to flash to my [Infinity 60%](https://www.massdrop.com/buy/infinity-keyboard-kit) though, not the Infinity Ergodox.

Depariel, 2016-08-08:
@finleye Yes, `infinity_chibios` is intended for the Infinity 60%, NOT the Infinity Ergodox.

Depariel, 2016-08-08:
@F1LT3R If you want to use QMK on your Infinity Ergodox, you should have a look at the [Ergodox](https://github.com/jackhumbert/qmk_firmware/tree/master/keyboards/ergodox) section.

fredizzimo, 2016-08-19:
I will rename infinity_chibios to infinity60 soon for less confiusion. And also update the readme with instructions for building.

Meanwhile the build instructions should be exactly the same as for the Ergodox, and described in the Ergodox readme. 

Are you still having problems with getting it to build? Also note that currently there's no support for the 1.1a version. But I can adapt the pull request from here https://github.com/tmk/tmk_keyboard/pull/371 to add support. I can't test it myself though.

Depariel, 2016-08-19:
There's support for 1.1a, just not for its LEDs. I added it in https://github.com/jackhumbert/qmk_firmware/pull/599 and have been running it on my 1.1a board ever since with zero issues.

closed 2017-04-19 by fredizzimo