qmk/qmk_firmware #604: Cannot get CTL_T(KC_LCBR) to work

Mauin, 2016-08-02:
While playing around with my keymap I'm currently facing issues with two key combinations.

`CTL_T(KC_LCBR)`
`ALT_T(KC_LABK)`
and their respective counterparts.

Basically I want my control key to type `{` when tapped and my ALT key to type `<` on tap

However even though I'm using the auto shifted versions that should type the correct characters, the keys are only typing the non-shifted variants of their keys, so control tapped = `[` and ALT tapped = `,`

Can you lead me into the right direction as to how to get this behavior to work?

kuel, 2016-08-09:
This issue is partly related to issue #511 because the workaround is similar.
This isn't a perfect solution, CTL and ATL are immediately registered once the key is pressed. I have tried to make it so they are only registered after the delay, but I can't get it to work.
The code is derived from keyboards/ergodox/keymaps/dvorak_spanish/keymap.c
```c
// MACROS, add these in place of CTL_L(KC_LCBR) and ALT_T(KC_LABK)
#define CLBRACE 0
#define ALTSIGN 1

static uint16_t key_timer;
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) { 
        case CLBRACE:
      if (record->event.pressed) {
        key_timer = timer_read();
        register_code(KC_LCTL);
      } else {
        unregister_code(KC_LCTL);
        if (timer_elapsed(key_timer) < 100) {
          register_code(KC_RSFT);
          register_code(KC_LBRC);
          unregister_code(KC_LBRC);
          unregister_code(KC_RSFT);
        }
      }
      break;
    case ALTSIGN:
      if (record->event.pressed) {
        key_timer = timer_read();
        register_code(KC_LALT);
      } else {
        unregister_code(KC_LALT);
        if (timer_elapsed(key_timer) < 100) {
          register_code(KC_RSFT);
          register_code(KC_COMM);
          unregister_code(KC_COMM);
          unregister_code(KC_RSFT);
        }
      }
      break;
  }
  return true;
}
```
The 100 delay can be set to your liking.

Xyverz, 2016-08-17:
I'm not sure whether my experience would be good to post here, but I can't even get KC_LCBR or RCBR to work at all. I have to use SFT(KC_LBRC) or RBRC instead. Is it possible that this is related to the issue you're seeing, or am I experiencing a whole different issue?

I'm using these key mappings on my raise/lower layers and not the main layers, so I don't know if that has anything to do with it.

closed 2017-02-09 by Mauin