qmk/qmk_firmware #606: Change SFT_T timer

tdroxler, 2016-08-03:
Hi everyone,

can I change the SFT_T timer? I've have  `SFT_T(KC_BSPC)` but I'm usually to fast and I erase instead of making a capital letter.

thanks

replicaJunction, 2016-08-17:
Someone who knows more may be able to give you a better solution, but I've been able to adjust this with the TAPPING_TERM variable. [Here's the TMK documentation on tapping](https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/doc/keymap.md#4-tapping), and I don't believe the implementation has changed much here.

Add this to your config.h:

`#define TAPPING_TERM 150`

This is the number of milliseconds you need to hold a key down before it's considered to be a "hold" keypress instead of a "tap" keypress. The default is 200ms.

tdroxler, 2016-08-25:
oh yeah, it works like a charm,

thank you

closed 2016-08-25 by jackhumbert