qmk/qmk_firmware #611: Keyboard does not register keys that are held down during power-up

0xdec, 2016-08-06:
I run several OSes on my machine and I need to press F11 or DEL during boot to access the boot menu or BIOS. If I power on my computer and immediately hold down any key, it won't actually register unless I press it again after the keyboard has turned on, so it's difficult to time it right so that the motherboard actually recognizes it. Could the default behavior be changed so that any keys are recognized immediately upon the firmware finishing its initialization? I took a quick look and it seems the answer may lie somewhere within matrix_init().

jackhumbert, 2016-08-08:
Ah, that's an interesting issue - do you have bootmagic off? That may be interfering with that. If it still doesn't work, this is definitely something I'd want to add (at least as an option).

0xdec, 2016-08-09:
Yeah, bootmagic is disabled.

drashna, 2018-03-25:
Is this still an issue? 

Also, yeah, adding this to your keymap.c may help:
```
void matrix_init_user(void) {
  register_code(KC_F11);
  wait_ms(3000);
  unregister_code(KC_F11); 
}
```

This should work and hold F11 down for the first 3 seconds of powering on.

0xdec, 2018-03-27:
This is still an issue as of 298ac18

adiabatic, 2018-04-03:
Still an issue, as is #1149 (probably the same issue).

fredizzimo, 2018-04-07:
After a very quick lock, I think this is the offender https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/lufa/lufa.c#L555

Try to comment that out, and see if it works better.

AGausmann, 2018-09-19:
I'm having a similar problem, I'm unable to use any QMK keyboard during the boot process, including the GRUB boot menu.

drashna, 2018-10-21:
@AGausmann have you tried commenting out the relevant like of code?

dylancopeland, 2019-04-14:
I am also running into this same issue. Has this been resolved in a recent firmware release?

For those also with this issue, if your keyboard has a backlight, enable it and wait for it to be illuminated before pressing the key. I've been doing that to resolve the timing issues.

aaahrens, 2019-10-23:
i am also getting this issue, any updates?

thealpa, 2020-11-07:
Still an issue as of https://github.com/qmk/qmk_firmware/commit/bd1c048a76e0ba84513caf1d59573dc55618c99a, I tried commenting out `clear_keyboard();` from https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/lufa/lufa.c#L623 but this doesn't resolve the issue for me.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc

awilkins, 2023-02-17:
Arrived here after discovering my Keychron Q6 doesn't work in GRUB, which means I have to plug in another keyboard to change boot options.

From reading the source, the firmware for this one uses `CUSTOM_MATRIX = lite` ... I don't know when this keyboard is "properly" initialized.

awilkins, 2023-04-13:
I found that upgrading my Q6 to the latest release firmware from Keychron fixed my boot-time issues.