qmk/qmk_firmware #625: Question: How to build a uk layout

am, 2016-08-12:
This may be a silly question but I could not found a way to achieve it. 

I'm using OSX, and I have 'Unicode Hex Input' selected on the OS. What I would like to achive is to have the same behaviour when pressing <kbd>alt</kbd> and <kbd>e</kbd> that produces a <kbd>´</kbd>.

I saw [this files](https://github.com/jackhumbert/qmk_firmware/tree/master/quantum/keymap_extras), but I could not found any example how to use it. I've tried to include it but I see no changes on the input.

iFreilicht, 2016-08-14:
Why would you want to do it with unicode hex input? Anyway, there are multiple ways to do this, but the easiest will be to change the `process_record_user` function in your `keymap.c`.

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    if (keycode == KC_E && keyboard_report->mods & (MOD_BIT(KC_LALT)))
    {
        process_unicode(UC_GRV, record);
        return false;
    }
    return true;
```

But you can also replace the `process_unicode` line with this:

```
register_code(KC_GRAVE);
unregister_code(KC_GRAVE);
```

That way, you won't need unicode hex input and it will work on all OS'.

am, 2016-09-02:
Thanks @iFreilicht for the help here. 

I use unicode since it's the only layout that does not mess with the configuration I defined in the firmware. But I still miss a way to input latin characters, that's what I'm trying to accomplish. 

Maybe there more elegant way to do it, in the past I was using an UK keyboard and I could get those chars using <kbd>alt+e</kbd> <kbd>alt+n</kbd> and so on...

Ideas are more than welcome 😄

algernon, 2016-09-02:
There are a number of ways you can do this: following @iFreilicht's suggestion of using `process_record_user` is one, but that requires quite a few lines of code, and `if` branches, or a `switch` statement. Another option would be to replace your `Alt` keys with macros, that register `alt`, and also switch to a layer where `e`, `n` and so on are replaced by `UC_` symbols. Thus, when you hit `alt+e`, you'll hit the `e` of the layer, and that'll get you the unicode char you configured for it.

Letting `alt` up would then deactivate the layer, and unregister `alt` itself.

Something like this:

```c
case M_LALT:
  if (record->event.pressed) {
    layer_on(_ALT);
    register_code(KC_LALT);
  } else {
    layer_off(_ALT);
    unregister_code(KC_LALT);
  }
  break;
```

And on the `_ALT` keymap, in the `e` position, you could have `UC(00E9)`, for example.

am, 2016-09-27:
Hi,
Thanks for the heads up. I've tried something similar to what @algernon proposed, I've created a new Layer to map those chars. The thing is that using Unicode Input when I type <kbd>alt+e</kbd> I get a question mark symbol, as the OS is not able to get that value, at this point I'm not sure if I really need some software support to be able to output latin chars.

I also read [the note in the readme about supporting Unicode](https://github.com/jackhumbert/qmk_firmware#unicode-support), I've enabled that to see if it fixes the problem, but didn't get any luck.

[This are the changes](https://github.com/am/qmk_firmware/commit/257f08f3ea000b1a9bfb0142ad79ae4cee38b8f1) that I'm currently compiling.

At some point I think I should change the title of the question, since what I try to achieve is not to have an UK layout, but rather how to output latin chars on Unicode layout.

Cheers

am, 2016-09-28:
Reading [this article](http://www.poynton.com/notes/misc/mac-unicode-hex-input.html) the origin of the problem might be the alt key being treated in a different manner when Unicode Hex Input is selected:

> In the normal, default U.S. or Canadian English situation, holding option down while tapping certain keys (e or i or u, for example) remembers an accent (such as ´ or ˆ or ¨ respectively) but makes no mark. The immediately following keystroke (typically a vowel) produces an accented letter (providing the particular letter keyed is capable of taking the particular accent).

(...)

> **Choosing the Unicode Hex Input method alters the behaviour of the “option” key.** To place U+03B1 into your document, hold down the option key and – while holding it down – tap the four keys 0, 3, b, and 1 in sequence. On pressing the last of the four, the character α appears; you can then release the option key. 

If this is the case, is there an option to get the behaviour of normal <kbd>alt</kbd> key?

slebedev, 2016-10-28:
I faced the same problem while trying to improve leader sequence that should print shruggie (as was done by @algernon [here](https://github.com/algernon/ergodox-layout/blob/master/keymap.c#L977..L985).

The result is `ººåƒ\_(£ºç¢)_/ººåƒ` while this one `¯\_(ツ)_/¯` is expected. =(

Moreover, if I bind, for example, `¯` unicode symbol in my keymap by placing `UC(0x00af)`, the result is `ººåƒ`...

am, 2016-10-28:
While I still have curiosity how to do this, my original problem was solved as soon I realise that US input has the keys in the exact same place as the Hex input. ¯\_(ツ)_/¯

I know this sounds noob, but from someone that came from UK layout it was not obvious since the beginning.

My own limitation does not exist anymore, but this question is still unanswered, I will leave the issue open.

slebedev, 2016-10-29:
It seems that I haven't read [this article](http://www.poynton.com/notes/misc/mac-unicode-hex-input.html) carefully. I read it again more closely and realized that correct input source wasn't enabled (Unicode Hex Input). =( Sorry my inattention, please. I have changed input source from US layout to Unicode Hex Input and now it works as expected. `¯\_(ツ)_/¯`

@am, thanks for link to that article!

closed 2017-06-27 by skullydazed