qmk/qmk_firmware #632: process_unicode() is not fit for Windows

iFreilicht, 2016-08-14:
I tried using `process_unicode()` today and it doesn't work with Windows at all (at least Windows 10, don't have any other versions to test on.

Looking at the code, there are multiple issues.

Firstly, Windows doesn't recognise hex bytecodes for Unicode, only decimal. So for Ä, the number isn't 00C4, but 0196.

Secondly, the number row can't be used for this input. You have to use the numberpad codes, otherwise ALT+0 for example can be recognised as a shortcut by the current program and will not yield a unicode character.

Thirdly, NumLock has to be activated for this to reliably work. I also tried it without, and that only works for some codes. For example, 0223 generates an ß, regardless of NumLock state, but 0196 only generates an Ä if NumLock is activated.

Can anybody else confirm that these are the restrictions in Windows?

algernon, 2016-08-14:
I'm working in the unicode area, will try it on my gaming PC sometime soon (Win 8.1), and report back with my findings.

The Linux unicode sequence stuff was wrong too, I'll be sending a pull request about that tomorrow or so.

iFreilicht, 2016-09-12:
Not necessarily. This key doesn't exist on my machine and I was able to input Unicode using the numberpad. I already found out why it doesn't work, did you read my post?

hia3, 2016-09-12:
If `EnableHexNumpad` does not already exist in your registry, you will need to create it by right clicking in the right panel and selecting "New | String Value".

DidierLoiseau, 2016-09-14:
With `EnableHexNumpad` enabled (though I don't remember whether it really made a difference), I have it working in Excel, Firefox and Notepad++ (and probably others).

However I could never make it work in Word, Outlook and other Rich Edit controls. In those, typing the hex code (without alt) and then pressing Alt+X works though. But this does not work anywhere else…

closed 2017-06-27 by skullydazed