qmk/qmk_firmware #636: Support for version 1.2 of KC60?

mkraemer, 2016-08-15:
The PCB of the KC60 i just received from Massdrop is version 1.2 which I haven't seen mentioned anywhere else..
Has anybody experience with using qmk firmware on a non-2.0 KC60?

Thanks :)

sgoodwin, 2016-08-15:
Can you post a picture of the PCB? Or perhaps check the pin-outs yourself? If the pins for the rows/columns match the 2.0, it should be fine.

mkraemer, 2016-08-16:
Thanks for your response!
I've attached pictures :)

> Or perhaps check the pin-outs yourself? If the pins for the rows/columns match the 2.0, it should be fine.

Can you point my in the right direction on how to do this?

What's the worst case scenario when flashing qmk to my maybe-not-fully-supported KC60? Non-working rows/column or a bricked and unflashable board?

Thanks again!

[image]
[image]
[image]

sgoodwin, 2016-08-16:
It's the same chip, so your only risk is that certain columns/rows might be in the wrong order. If you trace (like with a finger/something pointy) along the board there from the chip to a row, you'll see which pin is used for which row. You then want to compare that order to the ones used in the 2.0:

rows: { D0, D1, F6, F7, D5 }
columns { F0, F1, E6, C7, C6, B7, D4, B1, B0, B5, B4, D7, D6, B3 }

mkraemer, 2016-08-16:
Everything seems to be working great!

Awesome! Thank you very much. :+1:

closed 2016-08-16 by mkraemer