qmk/qmk_firmware #641: mkdir.exe has stopped working

Dr-Brains, 2016-08-16:
Hi,
I have my planck for about a week and wanted to customize the layout.
After having set up my build environment by following the steps in the readme, I tried to verify my installation by running the AVR shell and running the following command:

`C:\Users\Admin\Desktop\qmk\keyboards\planck>make`

When I run this command I get an error that mkdir.exe has stopped working.
If I don't close the command line, I get these errors:

```
C:\Users\Admin\Desktop\qmk\keyboards\planck>make
      0 [main] date 8936 open_stackdumpfile: Dumping stack trace to date.exe.stackdump
      0 [main] date 10032 handle_exceptions: Exception: STATUS_ACCESS_VIOLATION
    402 [main] date 10032 open_stackdumpfile: Dumping stack trace to date.exe.stackdump
/usr/bin/sh: cmp: command not found
/usr/bin/sh: ../../.build/obj_planck_rev4_default/cflags.txt: No such file or directory
../../tmk_core/rules.mk:438: recipe for target '../../.build/obj_planck_rev4_default/cflags.txt' failed
make: *** [../../.build/obj_planck_rev4_default/cflags.txt] Error 1
```
How can I resolve this problem?

PS: Don't know if it helps, but this is the content of the stackdump file:
```
MSYS-1.0.18 Build:2012-11-21 22:34
Exception: STATUS_ACCESS_VIOLATION at eip=60897244
eax=00000220 ebx=60EA022C ecx=0000000F edx=00000000 esi=00000220 edi=60EA022C
ebp=0061FEA0 esp=0061FE88 program=c:\Program Files (x86)\MHV AVR Tools\bin\date.exe
cs=0023 ds=002B es=002B fs=0053 gs=002B ss=002B
Stack trace:
Frame     Function  Args
0061FEA0  60897244  (60EA022C, 00000220, 0061FEF4, 60EA26C4)
0061FEC0  60808A3F  (60EA00D4, 00000220, 000000CC, 0061FEF4)
0061FF10  60804426  (6089C740, FFFFFFFE, 000000CC, 6089C6A4)
0061FF50  60804995  (00000000, 00000000, 00000000, 00000000)
0061FF70  0040A75F  (004014C0, 037F0000, 0061FF94, 75D162C4)
0061FF80  0040103E  (00228000, 75D162A0, D2355707, 0061FFDC)
0061FF94  75D162C4  (00228000, 1542A7D1, 00000000, 00000000)
0061FFDC  77EA0609  (FFFFFFFF, 77EC2514, 00000000, 00000000)
 119494 [main] date 7560 handle_exceptions: Exception: STATUS_ACCESS_VIOLATION
 121184 [main] date 7560 handle_exceptions: Error while dumping state (probably corrupted stack)
```

iFreilicht, 2016-08-16:
I've had this very same issue, and for me the solution was to install cygwin (a terminal emulator for windows) and run all building from there. It's the only terminal I could get this running in, but now it works buttery smooth.

You might have to execute the two setup scripts inside cygwin first, but after that it should work. 

If the installation of cygwin is a little confusing, just use the standard settings everywhere, can't do anything wrong with that.

Dr-Brains, 2016-08-16:
Aha, that does seem to do the trick! I did not have to execute the setup scripts though, it ran straight out of the box... Thank you very much!

closed 2016-08-16 by Dr-Brains