qmk/qmk_firmware #642: Cannot Use Shifted Keys with Modifier/Tap Shortcut Mappings

common-nighthawk, 2016-08-16:
`SFT_T(KC_UNDS)` does not work as expected.  when tapped, it sends dash (-) rather than underscore (_).

i want to dig into this myself more.  but i didn't see this reported as a bug.

presumably--because shift is an extra modifier to all the shortcut shifted keys, it's getting lost in this `MT(MOD_LSFT, kc)`.

algernon, 2016-08-16:
My suspicion is that `SFT_T` will call `register_code` under the hood, which only supports 8bit keycodes, and thus no modifiers can be applied there.

common-nighthawk, 2016-08-16:
thanks for such a quick response.  bummer for me.

is there any obvious workaround?  something like `MT(MOD_LSFT, KC_LSFT | KC_MINS)`
(bleh... this seems very unlikely.)

<sup>[out of scope of this ticket, but i've found https://asylum.madhouse-project.org/blog/tags/ergodox/ incredibly helpful digging into all this.  thank you!!!]</sup>

algernon, 2016-08-16:
The easiest way is likely a macro. When pressed, it starts a timer, and registers a `shift`. When released, it checks the timer, and if the elapsed time is smaller than `TAPPING_TERM`, it registers and unregisters `-` (and with `Shift` still being registered, that will turn into an underscore), and then unregisters `Shift` too.

Something like this:

```c
case M_SUNDS:
  if (record->event.pressed) {
    sunds_timer = timer_read();
    register_code (KC_LSFT);
  } else {
    if (timer_elapsed (sunds_timer) > TAPPING_TERM) {
      register_code (KC_MINS);
      unregister_code (KC_MINS);
    }
    unregister_code (KC_LSFT);
  }
  break;
```

And somewhere near the top of your keymap, define the `sunds_timer` variable like this: `static uint16_t sunds_timer`.

Not tested, but should work, if it compiles. The downside of this is that the underscore registers on keyup, you can't hold it to let the OS repeat it, but you wouldn't be able to do that anyway, because when held, the key acts as shift.

Hope this helps!

common-nighthawk, 2016-08-16:
this is amazing!  thank you so much--i really appreciate it.  i will make the updates/flash later today.

toneman77, 2016-08-24:
Just for the sake of completeness: I tried algernon's code and it works... 
...as soon as you change " > TAPPING_TERM" to " < TAPPING TERM".

g-sam, 2016-09-25:
an alternative quick fix is to look at where the space cadet feature is implemented in `./quantum/quantum.c` and just copy it but change the relevant keys to whatever you want (you also need to add your new keycode in `keymap.h`).

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @common-nighthawk please reopen this or open a new issue.

closed 2017-06-27 by skullydazed

ELLIOTTCABLE, 2020-01-17:
It'd be really nice if this were a built-in feature, honestly. )=

cattanimarco, 2021-07-15:
maybe a customisable space cadet?