qmk/qmk_firmware #649: keymap_common.h  - missing

nclundsten, 2016-08-18:
https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/readme.md#finding-the-keycodes-you-need

i dont see this file in /quantum/

Xyverz, 2016-08-18:
I can verify this. The file that has this information is named "keymap.h" and not "keymap-common.h"

fredizzimo, 2016-08-19:
I think we should rename the file to keymap_common, rather than fix the readme. The corresponding c files is also named keymap_common.h. That would require fixing all the existing keymaps though.

iFreilicht, 2016-08-20:
I don't think that is a good idea, every single keymap in QMK (and on all forks) includes `keymap.h`.  The c file is called `keymap_common.c` because you have to include the  `keymap.c` every time as well. In my eyes that naming makes more sense, and updating the readme is easier than updating all keymap files.

fredizzimo, 2016-08-20:
Actually **no** keymaps should include `keymap.h` directly. They only need to include the keyboard specific header file. And that's what the template does too.

Currently a couple of Ergodox keymaps and a few Planck ones do it, but no other keyboards are including it.

I would rather see some consistency here, because `keymap.h` is really the header file of `keymap_common.c`. `keymap_common.c` used to be named `keymap.c` but the problem with that is that we then have two object files named `keymap.o`, which is not a good idea, and can easily cause strange bugs.

I also think that the keycodes should be split into it's own header file for an easier reference. But that should probably be done at the same time as doing what I proposed in #513, in my last message. But even there I didn't take it as far as I really want. I will open a new issue, describing all the details at some point. But the basic idea would be the following.

I would really like to see that you could specify the OS keymap from Makefile or make commandline similar to this `make os_keymap=sv`. Which would automatically convert to the right OS layout. It's not as simple as switching out the keymap include file though, as the shift state of the characters are different on different layouts. So in some cases we would have to temporarily fake that shift is released for example.

iFreilicht, 2016-08-20:
Oh yeah right, the board header includes it. Sorry.

jackhumbert, 2016-08-23:
I'm a fan of switching `keymap.h` to `keymap_common.h` :) initially they were both `keymap` because quantum was overriding some of the old TMK keymap_common stuff, but now that's not needed!

jackhumbert, 2017-06-27:
I believe the initial issue here was resolved - @nclundsten, please reopen if I'm mistaken!

closed 2017-06-27 by jackhumbert