qmk/qmk_firmware #657: Advanced tap vs hold macro

kthibodeaux, 2016-08-18:
Case one and two below are incomplete.  I am not sure what I need to do to fill in the code where there are comments.  Any tips?

I'm aware I could use `CTL_T(KC_T)`.  I do currently ant the delay is a major issue.

Thank you for your time :)

```
static bool kt_controlling[2] = {0, 0};

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt)
{
  // MACRODOWN only works in this function
  switch(id) {
    case 0:
      if (record->event.pressed) {
        register_code(KC_LCTRL);
        register_code(KC_SPC);
        unregister_code(KC_SPC);
        unregister_code(KC_LCTRL);
      }
      break;
    case 1:
      if (record->event.pressed) {
        if(kt_controlling[1] == true) {
          register_code(KC_T);
        } else {
          kt_controlling[0] = true;
          register_code(KC_LCTL);
        }
      } else {
        if(kt_controlling[0] == true) {
          kt_controlling[0] = false;
          unregister_code(KC_LCTL);
        } else {
          unregister_code(KC_T);
        }
        // if nothing else was pressed act like T
      }
      break;
    case 2:
      if (record->event.pressed) {
        if(kt_controlling[0] == true) {
          register_code(KC_N);
        } else {
          kt_controlling[1] = true;
          register_code(KC_LCTL);
        }
      } else {
        if(kt_controlling[1] == true) {
          kt_controlling[1] = false;
          unregister_code(KC_LCTL);
        } else {
          unregister_code(KC_N);
        }
        // if nothing else was pressed act like N
      }
      break;
  }
  return MACRO_NONE;
};
```

algernon, 2016-08-19:
I'm not sure I understand the intent... what are you trying to achieve here? Can you describe the desired behaviour?

kthibodeaux, 2016-08-19:
I use T and N as CTRL keys (Colemak).  However there is a time delay for it to turn into control.  I would like to be able to remove that time delay.  At the same time I would like to disable T's ability to become control if N is already held down (this works by the kt_controlling variable).

The issue is that just tapping activates control (which I want), but if I release without pressing another key I would like the better to be typed.  Similar to the space cadet shifting.

algernon, 2016-08-19:
Ah, I see. So, a set of macros that register `CTRL` as soon as the key is pressed, but register as `T`/`N` if no other key was pressed while `CTRL` was held is what you're looking for? Additionally, if one is held, the other should act as the key, and not turn into control.

This is a tad complicated, but not impossible! We'll need to hook into `process_record_user()` too.

First, lets have a global where we can record what's up:

```c
static struct {
  bool pressed;
  bool fire_key;
} ctrl_state;
```

Then, we create the macros:

```c
case M_TCTRL:
  if (record->event.pressed) { // key pressed
    if (ctrl_state.pressed) { // a control is already pressed
      register_code(KC_T);
      ctrl_state.fire_key = false;
    } else { // a control is not yet pressed
      ctrl_state.pressed = true;
      register_code(KC_LCTL);
      ctrl_state.fire_key = true;
    }
  } else { // key released
    if (ctrl_state.pressed) { // a ctrl is already pressed
      ctrl_state.pressed = false;
      unregister_code(KC_LCTRL);
      if (ctrl_state.fire_key) {
        register_code(KC_T);
        unregister_code(KC_T);
      }
    } else {
      unregister_code(KC_T);
    }
  }
  break;
```

And the same for the `N` case, with the obvious changes.

We then add a `process_record_user` function, so we can register when a key is pressed while `CTRL` is held, and update our state:

```c
bool process_record_user (uint16_t keycode, keyrecord_t *record) {
  if (ctrl_state.pressed && (keycode != M(M_TCTRL)) && (keycode != M(M_NCTRL))) {
    ctrl_state.fire_key = false;
  }
  return true;
}
```

Lets look at the scenarios!

### Press `T` and release it, without any other key pressed

We enter the macro, register `CTRL`, set the pressed state, and register that on keyup, we shoud register `T`. Then, we release the key, enter the macro, notice that `CTRL` is pressed, and that `fire_key` is still set: it unregisters `CTRL`, and sends a `T`.

### Press `T`, hold it, press `C`

We enter the macro, and do our business. When `C` is pressed, we go through `process_record_user`, and since the `pressed` flag is set, we set `fire_key` to false, and proceed as usual - register `C`. Then we release both, and in case of `T`, we enter the macro. This time, however, `fire_key` is false, so we just unregister `CTRL`.

### Press and hold `T`, tap `N`

We enter the macro as usual for `T`. For `N`, we enter the macro, notice that `CTRL` is already pressed, so we register `N`, and set `fire_key` to false. When we release them, all is well too.

## Closing thoughts

All of this is untested, so may or may not work. But I hope it gets the idea across. Let me know how the code above fares!

kthibodeaux, 2016-08-19:
This code works well, but the side effects are worse than i thought they would be.  I really appreciate your help :D  I will try to change TAPPING_TERM and see what happens.

closed 2016-08-19 by kthibodeaux