qmk/qmk_firmware #667: Fix all the warnings and enable warnings as errors

fredizzimo, 2016-08-20:
I have noticed that we have got quite a few new warnings in the build logs recently.

As the warnings always indicates a potential problem, and almost always a real one, I suggest that we get them all fixed, and enable warnings as errors. We have some of the more annoying warnings disabled anyway so it shouldn't be a huge problem. 

Additionally for C a lot of the warnings should really be errors anyway, as you can easily call non-existing functions, which only generates a warning. Only to have the build fail with cryptic messages at link-time.

Also, even if the warnings could be hard to fix for a lot of our users, I think it's still better to have it immediately fail, than having the keyboard not work as intended when they actually try it out. 

Is anyone opposing the idea?

jackhumbert, 2016-08-23:
Yeah, I'm for this! As long as there is a way to easily disable it in the command line - sometimes I just need to quickly build/develop things without worrying about warnings.

fredizzimo, 2016-08-24:
Ok, so I guess the next step is to get all the existing warnings fixed, and then do the make file changes. I might have time to do that during the weekend, but if other people are faster then feel free to do it, especially for your own keymaps. 

It would be good if you write a comment here if you start working on it, so that I know and I will do so when I start too.

closed 2017-04-13 by fredizzimo