qmk/qmk_firmware #687: Enhancements to M-od, T-ap

linsomniac, 2016-08-26:
I have this idea that I'd like to implement, but I'm hoping someone can mentor me on it, I've never made code changes to the qmk firmware (though I started using C in the '80s).

I love the M-od, T-ap stuff, and use it for z and / and the key left of a and the ' keys...  But, I sometimes misfire because it is simply time based.  I feel like it could use some more smarts.

Here is my thinking, give me a sanity check on this:  Right now it works by a timer, if you release the key before the timer expires you get the tap, if not it is the modifier.  But, sometimes I press a key slowly and want a tap but get the modifier.  And conversely, sometimes I'm typing quickly and want a modifier, but do Ctrl-C so quickly that it instead sees /c...

My proposal: Have a few different timeouts depending on whether another key was hit with the M-T key:

- If I hit /c and quickly release /, say in 200ms, it sees it as Ctrl-C.

- If I hit / and release it in 200ms without any other key, it sees it as /.

- If I press / and then release it 2,000ms later, treat it as a modifier.  Maybe I am doing Ctrl to modify the mouse, or I think I want to type a Ctrl-C but decide I don't after thinking about it...

So the behavior would depend on whether another key was pressed, with a timeout above which it would always see the key as a modifier.  I feel like there should also be a low threshold below which it is always seen as two discrete keys.

Thoughts?  Pointers on where I should look to get started?

linsomniac, 2016-08-27:
Notes to self:

- Action is set up in quantum/keymap_common.c search for ACTION_MODS_TAP_KEY

- Action is ACT_MODS_TAP but is also used with MODS_ONESHOT and MODS_TAP_TOGGLE so I can't break those uses.

fredizzimo, 2016-08-28:
You might also be interested in these two Geekhack topics

https://geekhack.org/index.php?topic=41685.0
https://geekhack.org/index.php?topic=41685.0

I'm sorry that I can't give any more concrete advice at the moment, other than to try to use the new [Unit Test support](http://qmk.fm/#unit-testing) to help with the implementation.

algernon, 2016-08-28:
Look into tap-dance. Tap-dance came out of similar needs, and what you describe can be done with it, as far as I see. Otherwise a macro and a hook into `process_record_user` can do the trick for you:

- Have a macro, that starts a timer on keydown, and save that the key was pressed. If control is already registered, unregister it, and ignore the rest of this description.
- Hook into `process_record_user`, which runs every time a key is pressed.
- If in `process_record_user` you detect that another key was pressed before `/` reached keyup, register control immediately.
- In the macro, on keyup, check the timer, and see if control is already registered. If it is registered, and the timer is below 200ms, unregister it. If it isn't registered, and the timer is above 2000ms, register it. If control wasn't registered, and the timer is below 200ms, register and unregister `/`.

So, if you press `/c`, and `c` registers before `/`'s up, then the control gets registered before the `c`, due to the check in `process_record_user`. If you hit `/` and release it without any other key in 200ms, then the interrupt doesn't trigger, and you get `/`. And if you hold it, you get control on release, if you held it long.

If you want to get control while holding, then you will have to hook into `matrix_scan_user` too, and check the timer and the other state there too.

Hope this helps. In my opinion, you can do this entirely within your keymap, no need to modify anything in core QMK. I also think something similar can be done with tap-dance, but that's also a bit of code, and I'd need to think a bit more how to do that, so the above is the best I can offer at the moment.

closed 2017-06-27 by skullydazed