qmk/qmk_firmware #698: What does "KC_MSEL" mean?

SeanJM, 2016-08-28:
Hi, I am currently working on a front end for this tool and I want to design an icon for "KC_MSEL", for that to be clear, I would like to know what it means.

Proof of front end: https://twitter.com/SeanJMacIsaac/status/764432999983243264

jackhumbert, 2016-08-28:
[Media Select](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/keycode.h#L147) :) what it actually does, I'm not sure, but it likely varies per system.

SeanJM, 2016-08-28:
That is odd, I know it's 'Media Select'. I see you are as stumped as I am good sir. 💃

fredizzimo, 2016-08-28:
@tmk and @shelaf did some very nice investigation [here](https://github.com/tmk/tmk_keyboard/issues/332). 

So apparently it opens the media player in Windows 10 and does nothing on other platforms.

SeanJM, 2016-09-07:
Thank you!

Sean

closed 2017-06-27 by skullydazed