qmk/qmk_firmware #701: KC_ALT and KC_RALT does not works on planck

Dbroqua, 2016-08-29:
Hello,

I've a little bug on my Planck keyboard. Since few days KC_ALT and KC_RALT does not works with my layout or some other layouts.

If I replace KC_ALT and KC_RALT by others KEYS it's ok (not a PCB problem).

Have you the same issue with the actual revision of the project ?

Regards,
Damien

algernon, 2016-08-29:
Can you check if your alts register as the gui key instead? If so, you probably enabled the swap-alt-and-gui bootmagic option.

Dbroqua, 2016-08-29:
Hello @algernon ,

Thanks for your reply.

How to check this ? And how to reset this swap ?

algernon, 2016-08-29:
To check - what OS are you on? On Linux, `xinput` or `xev` can help, not sure about other OSes. If you can find a program where you can set keybindings by pressing the combo, try that, and press `ALT+A` for example, and see how the program recognises that.

To switch back, check the "Boot Magic Configuration - Virtual DIP Switch" section of `doc/TMK_README.md`: https://github.com/jackhumbert/qmk_firmware/blob/master/doc/TMK_README.md#boot-magic-configuration---virtual-dip-switch

Dbroqua, 2016-08-29:
I'm on GNU/Debian

I will try tonight at home.

Regards,
Damien

Dbroqua, 2016-08-29:
You'r in the right @algernon , ALT and GUI are swapped. But I can't revert it, I'm feeling a little bit stupid >< .

My process : 
- unplug keyboard
- press space, left, alt
- plug the keyboard

And... no change :(

I've also tried space+Backspace to clear the EEPROM but no change.

fredizzimo, 2016-08-29:
The bootmagic seems to be disabled by default on Planck. Try making your keymap, adding  `BOOTMAGIC_ENABLE=yes` to the make command.

I'm not sure how you have you have managed to get it set wrong in the first place though.

jackhumbert, 2016-08-29:
This is a feature in the [default keymap](https://github.com/jackhumbert/qmk_firmware/releases/download/v4.0/planck-keymap-v4.pdf) - pressing `Lower + Raise + g` should return this to normal. It's enabled by `Lower + Raise + h`.

Dbroqua, 2016-08-29:
Great !

Problem solved :D .

Thanks to all.

closed 2016-08-29 by Dbroqua