qmk/qmk_firmware #704: [Infinity ErgoDox] LCD backlight of slave stays lit

ickerwx, 2016-08-29:
I have flashed the latest release of QMK on my Infinity ErgoDox. The LCD backlight of the master half switches off after a short while, the slave backlight stays switched on. Can I actually interact with the LCD and/or the backlight or at least make sure both backlights are off?

fredizzimo, 2016-08-29:
Yes, I'm aware of that bug, and due to being to busy fixing other things I haven't been able to look into it yet. I'm actually planning to redo the whole LCD and LED support.

However that said, I think you should be able to switch both off by adding this to you make command, or add it to your keymap makefile. `LCD_ENABLE=no LED_ENABLE=no`. If it works, I can send a pull request that disables them globally.

ickerwx, 2016-08-30:
I tried these in my keymap Makefile as well as in the ergodox Makefile, but no success, the slave LCD backlight stays on. I think I'll live with it for now and keep my fingers crossed that you will deal with it when you have the time. Thanks a lot.

Zutatensuppe, 2016-10-06:
I have got the same problem. Also, is it possible to control the LCD at all with QMK?

ZibbeZabbe, 2016-11-20:
Any info on this? 
Also for fresizzmo's suggested fix do I add it to the make command like 
"make infinity-colemak LCD_ENABLE= no LED_ENABLE=no" 

Thanks for your time

fredizzimo, 2016-11-20:
I added a pull request #892, which completely disables the backlight. You need to flash both halves to fix the issue.

closed 2017-04-03 by ickerwx