qmk/qmk_firmware #708: KC_TILD not working with tap dance

apierz, 2016-08-31:
I am trying to get tap dance to work with KC_TILD so that when I double tap a key it outputs a ~. 

`
,[TD_G_TILD] = ACTION_TAP_DANCE_DOUBLE(KC_G, KC_TILD)
`

However there is no output when I use this in my keymap. I have several other tap dances set up and they are working without issue. 

I've also tried `,[TD_G_TILD] = ACTION_TAP_DANCE_DOUBLE(KC_G, LSFT(KC_GRV))` but get the same result.

algernon, 2016-08-31:
Are you using the master branch of QMK? Does `KC_TILD` or `LSFT(KC_GRV)` work on its own, without tap-dance?

apierz, 2016-08-31:
> Are you using the master branch of QMK?

Yes I cloned the master branch. 

> Does KC_TILD or LSFT(KC_GRV) work on its own, without tap-dance?

Yes, both work fine.

algernon, 2016-09-01:
Interesting, I'll have a look.

apierz, 2016-09-01:
Thanks! Please let me know if there's any tests I can run or information I can give you to help you out.

algernon, 2016-09-01:
I'll shout if there's anything I'd need help with. Didn't have time yet to try and reproduce the issue, though.

algernon, 2016-09-01:
Just tried it, and it works for me. When did you clone the master branch? Did you update it since? There was a recent-ish change, merged on August 18, that made tap-dance able to register keycodes with modifiers. Prior to that, modifiers were ignored at best, keys just didn't register at worst - which sounds a lot like your problem.

algernon, 2016-09-01:
Another idea, just to check something that keeps bugging me: if you pull `TD_J_TILDE` up, to be the first entry in the enum with a value of zero, does that fix the problem?

As far as I see in your keymap, you have other tap-dance keys that are shifted, and if they work, but the j/tilde one does not, then perhaps the bug is elsewhere - like too many tap-dance keys (though I doubt that would be the case).

Just a hunch, trying to rule out any possibilities.

apierz, 2016-09-01:
I pulled the latest commit from github, but no change

>if you pull TD_J_TILDE up, to be the first entry in the enum with a value of zero, does that fix the problem?

I tried this, but no change.

When I used an app to record key presses this was my output:
[image]
[image]

It looks like the shift is being registered but not the grave key.

apierz, 2016-09-02:
I wrote a standard function tap dance and, while a little clunkier, seems to work fine so far:

```
void g_tilde(qk_tap_dance_state_t *state, void *user_data) {
  switch (state->count) {
  case 1:
    register_code (KC_G);
    unregister_code (KC_G);
    break;
  case 2:
    register_code (KC_LSFT);
    register_code (KC_TILD);
    unregister_code (KC_TILD);
    unregister_code (KC_LSFT);
  }
}

```

`,[TD_G_TAPDANCE] = ACTION_TAP_DANCE_FN(g_tilde) 
`

When I left out the KC_LSFT it printed a `, even though I was using the KC_TILD argument.

algernon, 2016-09-02:
Yeah, `register_code` can't register shifted stuff. Can you try `register_code16(KC_TILD)` in the `case 2` branch?

apierz, 2016-09-02:
> Yeah, register_code can't register shifted stuff. Can you try register_code16(KC_TILD) in the case 2 branch?

I tried this and I got the same result as using KC_TILD in my original tap dance: the shift is registered but not the ~.

algernon, 2016-09-02:
Now that's very interesting, thanks for testing! Did you change the unregister too? (shouldn't matter, but to be safe...)

apierz, 2016-09-03:
Yes, I tried both.

xream, 2016-10-17:
I had the same issue before. However I can't reproduce it @80ac73a21cd08663d015b1fd16f34c6713f875d6.

closed 2017-06-27 by skullydazed