qmk/qmk_firmware #709: 'make' on the preonic

Truhammer, 2016-08-31:
running 'make' for the first time from within the \keyboards\preonic\keymaps\default folder.

windows error pops up saying date.exe has stopped working ..
along with some info about cmp command not dound and installing diffutils

here are a few screenshots.. the last is set path. Any help would be appreciated.

[image]
[image]

fredizzimo, 2016-09-01:
I think you are missing the msys package from Mingw, as the cmp command should be included in that. Try running`mingw-get install mssys`

The other problem is that the date command seems to be crashing, but it could be caused by the same thing.

mystcreater, 2016-09-02:
I have the exactly same problem and I haven't 2 days ago. Something have changed since that moment and cause the problem. By chance, I still have my code from 2 days ago and I'm still able to compile my hex file.

fredizzimo, 2016-09-02:
I think it's the makefile overhaul. The cmp command wasn't required before that.

However it didn't work correctly before since the quick target was removed and the makefiles were made to properly recompile the files that need to be recompiled due to makefile variable changes for example.

However the cmp command do exist in the mingw mssys package, so I'm pretty sure you don't have that properly installed somehow.

mystcreater, 2016-09-02:
Thanks for your fast answer :) 

I tried to execute the command that you give (mingw-get install mssys) but I receive a message indicating that the package mssys don't exist. Do you know what is missing in my computer?

fredizzimo, 2016-09-02:
Ah sorry the package is named msys and not mssys.

Truhammer, 2016-09-02:
I think it's mingw-get install msys

I think :)

mystcreater, 2016-09-02:
Thanks!! Very appreciated :)

The compilation works great but why the hex file is now at the root of the project instead of in the project where I compile the file? I use ergodox project and I build the file using the MakeFile in this project but now, my hex file is located at the root of the qmk firmware project? Is it a new behavior?

fredizzimo, 2016-09-02:
@mystcreater, I think that might have been broken with the big makefile overhaul as well.

I already opened a related discussion in #670, so I think rather than fixing it so it retains the old behaviour I will fix it according to that. So going with @jackhumbert's suggestion it would mean putting the output in the keymap folder.

But at the same time, we should probably ensure that the hex file is deleted from the current folder. Otherwise people might think that they are flashing a newly built hex file, while in fact they are flashing the old one that was left over.

I also think I will add some message at the very end of the compilation telling where it actually outputs the hex file.

mystcreater, 2016-09-02:
I just read the discussion #670 and your suggestion looks good. When you say that the hex will be in the keymap folder, do you mean at the same place where is the keymap.cs file? If it's right, the the first place that I've looked the first time that I've build my hex file 3 months ago.

fredizzimo, 2016-09-02:
Yes, that's right, the same folder as the `keymap.c` file

closed 2017-06-27 by skullydazed