qmk/qmk_firmware #712: 80-keys ergodox support

gaedol, 2016-09-01:
Hi, 
does this firmware support the 80 keys version of the ergodox?

I am trying to implement it, but I am failing to understand, inside ez.h which k-values I should add before and after k54 and k59 (respectively) to add support for those switches (the 4 more wrt the "standard" 76). 

Any help is appreciated!

algernon, 2016-09-01:
FWIW, I tried comparing the values for the EZ with those in benblazak's firmware, but the numbers there are quite different. I found a commit that changed the matrix there, and the old numbers were closer to what we have in the EZ, but I still have to look at it some more.

The PCB - according to various googled images - appears to have kc15, kc16 and kc17, kc18 for each hand, respectively.

gaedol, 2016-09-02:
This is actually solved in profet23's pull request.

closed 2017-06-27 by skullydazed