qmk/qmk_firmware #713: Cannot Compile with Tap Dance

common-nighthawk, 2016-09-02:
i get the following error--
```

keyboards/ergodox/keymaps/common-nighthawk/keymap.c:37:1: warning: implicit declaration of function 'ACTION_TAP_DANCE_DOUBLE' [-Wimplicit-function-declaration]
 [BASE] = KEYMAP(  // layer 0 : default
 ^
keyboards/ergodox/keymaps/common-nighthawk/keymap.c:37:1: error: initializer element is not constant
keyboards/ergodox/keymaps/common-nighthawk/keymap.c:37:1: error: (near initialization for 'keymaps[0][0][3]')
keyboards/ergodox/keymaps/common-nighthawk/keymap.c:37:1: error: initializer element is not constant
keyboards/ergodox/keymaps/common-nighthawk/keymap.c:37:1: error: (near initialization for 'keymaps[0][13][3]')
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_ergodox_ez_common-nighthawk/keyboards/ergodox/keymaps/common-nighthawk/keymap.o] Error 1
make: *** [common-nighthawk] Error 1
```

i added the following to my key map: `ACTION_TAP_DANCE_DOUBLE(KC_LSFT, KC_LBRC)`

i do not have a Makefile or config.h in my keymap directory.  therefore, i added `TAP_DANCE_ENABLE = yes` to qmk_firmware/keyboards/ergodox/Makefile so that it is 

```
SUBPROJECT_DEFAULT = ez
TAP_DANCE_ENABLE = yes

ifndef MAKEFILE_INCLUDED
	include ../../Makefile
endif
```

and i see that `#define TAPPING_TERM    200` is already set in qmk_firmware/keyboards/ergodox/config.h

lastly--i made sure i was current with the upstream.

algernon, 2016-09-02:
Can you show me the includes of your keymap?

common-nighthawk, 2016-09-02:
```
#include "ergodox.h"
#include "debug.h"
#include "action_layer.h"
#include "version.h"
```

algernon, 2016-09-02:
Ok, had a look - you're using tap-dance completely wrong. You don't put the `ACTION_TAP_DANCE_DOUBLE` in the keymap, but use it in the `tap_dance_actions` declaration. I'd recommend following the [docs](https://github.com/jackhumbert/qmk_firmware#examples), which have step-by-step guides on how to add this functionality to your layer.

Also, having a `Makefile` in your keymap dir will make a lot of things easier. These few lines will do:

```make
TAP_DANCE_ENABLE = yes

ifndef QUANTUM_DIR
	include ../../../../Makefile
endif
```

common-nighthawk, 2016-09-02:
sorry about that.  im in a bit over my head writing C.  thanks for all your help, @algernon--i now have the tap dance and macro features working nicely!!  which is tremendously cool.

ill close this issue, as the documentation is clear and i just misunderstood.

closed 2016-09-02 by common-nighthawk