qmk/qmk_firmware #714: Is it possible to gather usage data?

naps62, 2016-09-02:
I have an Ergodox EZ, and from time to time I change a couple of my keys (those that I already know I'm not using ideally, and that I want to gradually improve on)

One thing I would really love to have is statistics on my overall usage of the keyboard. This is not just knowing which keycodes I generate the most (which I could via the computer itself), but known which keys I actually use.

What I want, in practice is one of these 2:
1. Patch the firmware to store this usage data, and retrieve it sometime in the future. Is this even possible? Does the ergodox have any kind of memory I can use for this?
2. Some way of passing on this information to the computer, so I can store it there.

All I want is some pointers to any existing resources that might already exist about this, as I'm ok with patching the firmware myself to try and achieve this.

One practical example: I currently have two spacebar keys, because when I first got the keyboard I couldn't figure out on which hand I wanted it. Months went on, and when I think about it, I can't tell with 100% certainty which one I use the most, because it's not something I even notice while typing. So I would like to measure it.

I would also like to analyze my typing speed and patterns, to try to figure out which keys are hidering me the most, and where to optimize next.

algernon, 2016-09-02:
The easiest thing you can do to achieve this is to add a keylogger to your firmware, that logs to the HID console, and use `hid_listen` to collect it. Then you can use any other program to analyze the data.

I did something similar in [my own keymap](https://github.com/algernon/ergodox-layout#tools), and I build a heatmap out of the data, and also echo some stats every 500 keys or so about hand and finger usage across the board.

naps62, 2016-09-02:
@algernon thanks! thanks exactly what I want to do as well. I'll check out that keylogger, as well as your  keymap

cbbrowne, 2016-09-07:
Another approach to reporting it would be to set up a keystroke that runs a function that does a dump of this data.  I have an instance of this sort of thing where there's a keystroke I set up to dump out present version number (e.g. - planck/cbbrowne@gh60-0.1-2706-g73fdd8-dirty@2016-09-07-13:13:53).
In effect, you collect data into variables in the firmware, and then set up a function that calls SEND_STRING() to dump that as a series of keystrokes, and attach that function to some key mapping.
I'm not sure that's easier, though.

algernon, 2016-09-07:
Collecting the data in the firmware also increases the size of it considerably, and dumping it out as keystrokes is slooooooooooooooooow. I tried it before going the keylogger route. :)

naps62, 2016-11-01:
I have this mostly finished, based on @algernon's layout, so this can be marked as solved. Thanks so much!

closed 2016-11-01 by naps62

Lenbok, 2018-08-22:
@naps62 Can you share your code for this please?

naps62, 2018-08-22:
@Lenbok sure. I should have done that in the first place :)

My layout repo: https://github.com/naps62/ergodox-layout
And the heatmap generator: https://github.com/naps62/ergodox-heatmap-generator

I haven't used it in a while to be honest, because I had to format my system, and haven't gotten around to set this up again yet. I remember the heatmap generator still required a bit of manual work to get running (it was my first time messing with systemd)

Lenbok, 2018-08-22:
@naps62 Awesome thanks!! I'm looking at moving some of my modifiers (in particular ctrl, which as an emacs user I use pretty heavily, currently where caps lock normally sits, down to the thumb cluster on my redox) and got wondering about current thumb cluster utilization.

kdb424, 2018-08-22:
Two questions. First, is why not use whatpulse? Second, wouldn't this easily overblow the memory on the micro controller, or you can get away with it as it's arm and not as weak as the pro micro. Either way, that's awesome. Thanks for the share. Would love to see this an an option in QMK in the docs.

afinne, 2018-08-22:
This would be soo nice to have in QMK. I have used Whatpulse earlier to see which keys on my regular keyboard were used the most (and thus most likely causing discomfort https://asmand.wordpress.com/2015/12/07/battling-rsi-part-2/).
I have an Iris now, and it still is somewhat helpful to analyse keypresses as long as you don't for instance have multiples of the same button, but something in QMK would be really nice.

naps62, 2018-08-22:
@kdb424 I didn't even know about Whatpulse before (so thank you for the reference), but it wouldn't work for my use case.
To add to @algernon's answer, I mostly wanted this to gather metrics of my keyboard usage, independently of what the OS interpretation of those keys are. A couple of examples:

1. The original use case was that I started my ergodox layout by having two space keys, one on each side. I didn't know which side I'd want it in, and I honestly couldn't tell which one I was using the most even after months of using the keyboard (while typing, I'm on autopilot, not paying attention to those details). After measuring for a few weeks I was able to remove the one that was barely used.
2. I have a macro set for a key which triggers the shortcut to change language on my Linux setup. So I press a single key, and a `Super+Space` signal is sent to my OS. I want to measure how much I use that particular key, not how many Super+Space signals reached the OS