qmk/qmk_firmware #720: [Infinity ErgoDox] How to enable backlight support?

ickerwx, 2016-09-04:
I have ```BACKLIGHT_ENABLE = yes``` set in my keymap's Makefile and have a key that essentially calls ```backlight_step()``` when pressed. The code is copied from my Preonic keymap where it works like it should. Still, when I press the respective key on my Infinity Ergodox the backlight stays off.

I flashed the Kiibohd firmware, the LEDs light up like they should here, so it's not a hardware issue.
Do I have to do anything else beside setting ```BACKLIGHT_ENABLE``` in the Makefile and calling ```backlight_step()```?

fredizzimo, 2016-09-04:
Hi @ickerwx,

Yes, the Infinity Ergodox backlight support is not properly integrated into the QMK firmware yet. 

In my [TMK fork](https://github.com/fredizzimo/infinity_ergodox), which I made before moving over to QMK, things worked a bit differently. Although even there I didn't have support for dynamically changing the backlight. 

I could have got the same system working for QMK quite quickly, but I didn't want to do that. Instead I have been thinking about something that would work well with all kinds of keyboards and described in #717. 

The progress towards that have been quite slow, because first I needed proper unit test support, which in turn required some bigger changes to the makefile system. However now that those two things are done, I will finally have the time to start fixing the LCD and backlight support for the Infinity Ergodox. 

So I hope that we will have something working pretty soon.

ickerwx, 2016-09-04:
Alright, thanks for the info. I wasn't aware how much of the Infinitx Ergodox support is your work, thanks a lot for this, you're awesome.

b4lthazar, 2017-01-01:
Hey buddy, any news on this? I'd love to program the backlights and LCD screens :).

ec1oud, 2017-02-01:
keyboards/ergodox/keymaps/swissgerman/keymap.c shows one way to do it.

However last July I somehow had LCD graphics too; for some reason I don't see any keymaps doing that right now.  Nor setting the key backlighting, either.  I just got around to pulling the latest code from this repo, to see if any bugs got fixed and to try to sync up the keymaps again between work and home.

What's the holdup to getting all the features working again?  What will be the right way, to have an infinity-specific matrix_scan_user() function which does whatever we like, or will there be a way to customize the keymap and the graphics, backlighting etc. separately?  What should I do in the meantime to get it working again?

b4lthazar, 2017-02-03:
@ec1oud Nice. Can you please elaborate on the "hacks from last summer" ? And if I were to clone your fork, what other changes have you made? Thank you sir!

ec1oud, 2017-02-03:
It was based on some sample visualization code from fredizzimo.  The LCD has text showing which layer you are on, and the LCD backlight color changes depending on the layer, and I just want to have my keys illuminated so I didn't retain fredizzimo's visualization which put those backlight LEDs in motion, but the startup_animation is there, commented out.  The code is in my keymap: keyboards/ergodox/keymaps/ecloud-qwerty, so you'll probably have to port such changes to your own favorite layout, unless you like mine.  ;-)

Other changes you probably don't need are the previous patches where I was modifying the default OS X keymap (that was from before the subproject restructuring, and before there were so many keymaps to choose from).  But if you aren't building with that keymap, they won't matter.

I could make it less of a mess, but figured I'd see what architecture fredizzimo is going to come up with.

jH4nn3s, 2017-02-05:
Hey you guys,
I'm using @ec1oud 's tip/code and now want to adjust the visualizer.c file in the display_welcome method. In particular, I would like to display .bmp files. According to the documentation at https://wiki.ugfx.io/index.php/Images this is possible with ugfx and they provide example code. In this example code, 3 methods are called to actually open, draw and close the bmp, such as gdispImageOpenFile etc.
In the ugfx example code, only gfx.h is included. Since gfx.h is already included in the visualizer code and functions like gdispDrawString can be freely called in the visualizer.c, I assumed these image functions might be available too. 
However, upon compiling my adaptation of the example code, I get the error message "undefined reference to `gdispImageOpenFile'". I admit to being somewhat of a dilettante with C and header files. Does anybody know why the compiler does not find these ugfx Image functions or from where these would have to be included?

fredizzimo, 2017-04-19:
The LCD support is now merged and working. Not using the effect system that I was planning to do, but using the old visualizer code. The effect system is still on my roadmap, but it won't be too different from what we have now, it will just make things easier to use.

So I'm closing this.

closed 2017-04-19 by fredizzimo