qmk/qmk_firmware #721: dynamic macros

jceaser, 2016-09-04:
I'm trying to get the dynamic macro code to work. Following directions at:

https://github.com/jackhumbert/qmk_firmware/blob/bade8be66c3440cc91db1d56d3a29a5b5af226dc/readme.md#dynamic-macros-record-and-replay-macros-in-runtime

but to no avail. My issues seams to be the following code:

    bool process_record_user(uint16_t keycode, keyrecord_t *record)
    {
        /*if (!process_record_dynamic_macro(keycode, record))
        {
            return false;
        }*/
        return true;
    }

If I uncomment out that section, my keyboard's lights blink and emit a continuous tone. I can get the keyboard in flash mode as normal, but I feel as if there is some infinite loop or something the code has gotten itself into. Has anyone else worked with this code?

My code:

https://github.com/jceaser/qmk_firmware/blob/baltimore/keyboards/planck/keymaps/baltimore/keymap.c

Vifon, 2016-09-14:
Gotcha! Add a line like [this one](https://github.com/jceaser/qmk_firmware/blob/a17e59b285f1ce875aad0c85b93da7e57e8b4b65/quantum/dynamic_macro.h#L12) just before including the `dynamic_macro.h` header. I used 128 bytes and it started properly. So it would be:

```c
#define DYNAMIC_MACRO_SIZE 128
#include "dynamic_macro.h"
```

Your keymap is more memory hungry than mine so after reserving 256 bytes for the macro buffer, there was not enough left for the keyboard to actually work.

@jackhumbert, could you suggest a safe default value? It seems taking 256 bytes is definitely too much.

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @jceaser please reopen this or open a new issue.

closed 2017-06-27 by skullydazed

jceaser, 2017-06-27:
Yea I would like this open because I can not confirm that the code works as described