qmk/qmk_firmware #726: Console + HID Listen no longer works

IBNobody, 2016-09-05:
I have enabled console out via my makefile, have made sure that no_debug and no_print are not defined, and have started hid_listen.exe. I no longer see my keyboard connecting with hid_listen. 

Older builds appear to work fine.

fredizzimo, 2016-09-05:
Do you have any idea of what version you were using before it stopped working?

IBNobody, 2016-09-05:
#372

It was before the keyboard -> keyboards change.

IBNobody, 2016-09-05:
Specifically, the behavior of HID Listen was that it used to show the following...

```
Waiting for new device:.
Listening:

Device disconnected.
```

Now it is stuck in...

`Waiting for new device:........`

I suspect we killed something in the way the device connects to HID Listen.

fredizzimo, 2016-09-05:
Oh, that's a long time ago. So it's really hard to guess which changelist it could be.

But my initial suspect would be #677, the addition of a virtual serial port, which changes the USB descriptors.

IBNobody, 2016-09-05:
It was broken before that. I just tried #668 which was merged two days prior to #677 and was seeing the same behavior.

fredizzimo, 2016-09-05:
Ok, #536, has to work as it added documentation for it. #572 also most likely works.

My own evil #666, the makefile overhaul is always a suspecting pull request.

But other than that I don't see anything obvious.

IBNobody, 2016-09-05:
#536 didn't work. 

I am still trying to look backwards. I am running into problems because some of the commits I am trying to checkout return a  `fatal: reference is not a tree: ` error. Am I doing something wrong?

fredizzimo, 2016-09-05:
I think that pull request was made during the time when, pull requests were squashed. 

So you need to checkout the actual commit https://github.com/jackhumbert/qmk_firmware/commit/a69d002506d822fef9713b844cd512cefcffd4fc

algernon, 2016-09-05:
FWIW, console and hid_listen work for me on the ErgoDox EZ, with master as of last night, so the evil PR may not be the root cause.

IBNobody, 2016-09-05:
Hmm... Did something change to make it so that console had to be enabled in order for hid listen to connect? In my last working build of ArrowPad (that I am using to track down the cause of this issue), I had console disabled but still got hid_listen to connect.

Also, can someone take a look at my Planck keymap  (pvc) to see if there is a smoking gun? I turned off audio and turned on console in my local keymap makefile. That should be all I need for debug, right?

fredizzimo, 2016-09-05:
The ArrowPad enables the console by default. But the keymaps 21 and 24 turns it off. There might also have been some bugs in the make system that made it not disabling even if it was requested.

As far as I know the console have always been required.

I don't see anything wrong with your Planck keymap.

IBNobody, 2016-09-05:
Oh you'd know if it didn't get disabled. It would balloon up the code by 2k.

The cause of the issue is in #395 (d9e4dad0a828a8a904f44dda090a4d6d08fe2948). #364 (a5d638ad3091e2d553eaa04d933df28eeced5a8b) appears to work, and the very next merger was #395 which does not.

IBNobody, 2016-09-05:
@algernon I could try loading your build onto my Planck just to see if I can get a connection, I don't care if the keys don't work.

algernon, 2016-09-05:
Not sure that'd work, to be honest.

IBNobody, 2016-09-06:
Here are some more findings.

I am on the current branch.

If I run `make` from my arrow_pad directory, I get 3 hex files made. Only the default hex file has console turned on. When I program using the default hex, I get a `Listening: done. Keyboard start.` prompt in hid_listen. When I program the other two hex files, I do not get any response in hid_listen.

If I run `make default` and program the resulting hexfile, I get a  prompt in hid_listen.

If I change the console enabled line in my makefile for pad_24 from no to yes and run `make pad_24`, I get no listening response.

If I add `#define CONSOLE_ENABLE` to my keymap, or I force tmk_core\common.mk to be...

```# ifeq ($(strip $(CONSOLE_ENABLE)), yes)
    TMK_COMMON_DEFS += -DCONSOLE_ENABLE
# else
#     TMK_COMMON_DEFS += -DNO_PRINT
#     TMK_COMMON_DEFS += -DNO_DEBUG
# endif
```

I still get no console on pad_24.

IBNobody, 2016-09-06:
It appears to be related to defines. I was able to get console out on pad_24 with the following config.h...

```
BOOTMAGIC_ENABLE = yes # Virtual DIP switch configuration(+1000)
MOUSEKEY_ENABLE  = yes # Mouse keys(+4700)
EXTRAKEY_ENABLE  = yes # Audio control and System control(+450)
CONSOLE_ENABLE   = yes # Console for debug(+400)
COMMAND_ENABLE   = yes # Commands for debug and configuration
NKRO_ENABLE      = yes # Nkey Rollover - if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
BACKLIGHT_ENABLE = yes # Enable keyboard backlight functionality
MIDI_ENABLE      = no  # MIDI controls
AUDIO_ENABLE     = no  # Audio output on port C6
UNICODE_ENABLE   = no  # Unicode
BLUETOOTH_ENABLE = no  # Enable Bluetooth with the Adafruit EZ-Key HID
RGBLIGHT_ENABLE  = no  # Enable WS2812 RGB underlight.  Do not enable this with audio at the same time.

# Do not enable SLEEP_LED_ENABLE. it uses the same timer as BACKLIGHT_ENABLE
SLEEP_LED_ENABLE = no  # Breathing sleep LED during USB suspend

CONFIG_H = keymaps/$(KEYMAP)/config.h
```

`MOUSEKEY_ENABLE  = yes` and `EXTRAKEY_ENABLE  = yes` must both be set to yes in order for console to be enabled. The likely subject, tmk_core\common.mk looks good, though.

```

ifeq ($(strip $(MOUSEKEY_ENABLE)), yes)
    TMK_COMMON_SRC += $(COMMON_DIR)/mousekey.c
    TMK_COMMON_DEFS += -DMOUSEKEY_ENABLE
    TMK_COMMON_DEFS += -DMOUSE_ENABLE
endif

ifeq ($(strip $(EXTRAKEY_ENABLE)), yes)
    TMK_COMMON_DEFS += -DEXTRAKEY_ENABLE
endif

ifeq ($(strip $(CONSOLE_ENABLE)), yes)
    TMK_COMMON_DEFS += -DCONSOLE_ENABLE
else
    TMK_COMMON_DEFS += -DNO_PRINT
    TMK_COMMON_DEFS += -DNO_DEBUG
endif

```

Incidentally, I wonder if this is an issue related to #725.

IBNobody, 2016-09-06:
@fredizzimo It may be a descriptor problem. I confirmed that the console code was being included. 

I also saw that in the Windows Device Manager, an unknown mouse was being detected...

[image]

That matches the console device descriptor.

```
#ifdef CONSOLE_ENABLE
const USB_Descriptor_HIDReport_Datatype_t PROGMEM ConsoleReport[] =
{
    HID_RI_USAGE_PAGE(16, 0xFF31), /* Vendor Page(PJRC Teensy compatible) */
    HID_RI_USAGE(8, 0x74), /* Vendor Usage(PJRC Teensy compatible) */
    HID_RI_COLLECTION(8, 0x01), /* Application */
        HID_RI_USAGE(8, 0x75), /* Vendor Usage 0x75 */
        HID_RI_LOGICAL_MINIMUM(8, 0x00),
        HID_RI_LOGICAL_MAXIMUM(16, 0x00FF),
        HID_RI_REPORT_COUNT(8, CONSOLE_EPSIZE),
        HID_RI_REPORT_SIZE(8, 0x08),
        HID_RI_INPUT(8, HID_IOF_DATA | HID_IOF_VARIABLE | HID_IOF_ABSOLUTE),
        HID_RI_USAGE(8, 0x76), /* Vendor Usage 0x76 */
        HID_RI_LOGICAL_MINIMUM(8, 0x00),
        HID_RI_LOGICAL_MAXIMUM(16, 0x00FF),
        HID_RI_REPORT_COUNT(8, CONSOLE_EPSIZE),
        HID_RI_REPORT_SIZE(8, 0x08),
        HID_RI_OUTPUT(8, HID_IOF_DATA | HID_IOF_VARIABLE | HID_IOF_ABSOLUTE | HID_IOF_NON_VOLATILE),
    HID_RI_END_COLLECTION(0),
};
#endif

```

I am completely in the dark regarding USB beyond PID and VID, so I am not sure how much help I can be. :(

IBNobody, 2016-09-06:
And... Solved. The issue was that the USB descriptors changed enough to require a complete uninstall of the USB device. Previously, I could get away with removing ghosted/non-present keyboards, but apparently this was not enough.

1, Open Device Manager.
2. Windows - View - Devices By Type.
3. Identify the Keyboard (USB composite device).
4. Right-Click, Uninstall.

This isn't the first time I've been burned by this, sadly enough. You'd think I would have learned.

closed 2016-09-06 by IBNobody