qmk/qmk_firmware #745: Keycodes for õüöä

mihkell, 2016-09-11:
What are keycodes for there letters: õ ü ö ä?

coderkun, 2016-09-11:
Depends on your language setting, I guess. The `keymap_german.h` defines these e. g.:

```
#define DE_AE KC_QUOT
#define DE_UE KC_LBRC
#define DE_OE KC_SCLN
```

algernon, 2016-09-11:
These are - as coderkun above said - entirely defined by the layout in your OS. Some layouts have them available without modifers, some don't. If you have a Compose key (AltGr on US-intl, I believe), you can input `ü` with `AltGr + ", u` - which means that on US-intl layouts, you'll have to use a macro.

If your OS has a layout where these keys are readily available, then telling us your OS layout will help figuring out the keycode for the symbols.

mihkell, 2016-09-11:
Well I have OS X with US mapping. And previously I remapped with the keys ukelele so that Right_CMD+] was õ and so on. now I want Right_CMD+P to be õ and so on.
Hoped that maybe I can put it into to ErgoDox so I could move from computer to computer with ease :)

coderkun, 2016-09-11:
@mihkell then maybe using Unicode characters (e. g. `uc(0x00F5)`) might be an option.

sgoodwin, 2016-09-11:
If you're on OS X you can hold down the u o, a, or e key to get accented versions.

mihkell, 2016-09-11:
@coderkun good idea but it will give me "ººƒ§" atm. for UC(0x00F5) and UC(0xF5)

@sgoodwin just holding them down doesn't help me.

mecanogrh, 2016-09-11:
I use these in custom unicode.h file
`
#ifndef UNICODE_H
#define UNICODE_H

#include "quantum.h"
#include <math.h>

#define UC_BSPC	UC(0x0008)

#define UC_SPC	UC(0x0020)

#define UC_EXLM	UC(0x0021)
#define UC_DQUT	UC(0x0022)
#define UC_HASH	UC(0x0023)
#define UC_DLR	UC(0x0024)
#define UC_PERC	UC(0x0025)
#define UC_AMPR	UC(0x0026)
#define UC_QUOT	UC(0x0027)
#define UC_LPRN	UC(0x0028)
#define UC_RPRN	UC(0x0029)
#define UC_ASTR	UC(0x002A)
#define UC_PLUS	UC(0x002B)
#define UC_COMM	UC(0x002C)
#define UC_DASH	UC(0x002D)
#define UC_DOT	UC(0x002E)
#define UC_SLSH	UC(0x002F)

#define UC_0	UC(0x0030)
#define UC_1	UC(0x0031)
#define UC_2	UC(0x0032)
#define UC_3	UC(0x0033)
#define UC_4	UC(0x0034)
#define UC_5	UC(0x0035)
#define UC_6	UC(0x0036)
#define UC_7	UC(0x0037)
#define UC_8	UC(0x0038)
#define UC_9	UC(0x0039)

#define UC_COLN UC(0x003A)
#define UC_SCLN UC(0x003B)
#define UC_LT	UC(0x003C)
#define UC_EQL	UC(0x003D)
#define UC_GT	UC(0x003E)
#define UC_QUES	UC(0x003F)
#define UC_AT 	UC(0x0040)

#define UC_A 	UC(0x0041)
#define UC_B 	UC(0x0042)
#define UC_C 	UC(0x0043)
#define UC_D 	UC(0x0044)
#define UC_E 	UC(0x0045)
#define UC_F 	UC(0x0046)
#define UC_G 	UC(0x0047)
#define UC_H 	UC(0x0048)
#define UC_I 	UC(0x0049)
#define UC_J 	UC(0x004A)
#define UC_K 	UC(0x004B)
#define UC_L 	UC(0x004C)
#define UC_M 	UC(0x004D)
#define UC_N 	UC(0x004E)
#define UC_O 	UC(0x004F)
#define UC_P 	UC(0x0050)
#define UC_Q 	UC(0x0051)
#define UC_R 	UC(0x0052)
#define UC_S 	UC(0x0053)
#define UC_T 	UC(0x0054)
#define UC_U 	UC(0x0055)
#define UC_V 	UC(0x0056)
#define UC_W 	UC(0x0057)
#define UC_X 	UC(0x0058)
#define UC_Y 	UC(0x0059)
#define UC_Z 	UC(0x005A)

#define UC_LBRC	UC(0x005B)
#define UC_BSLS	UC(0x005C)
#define UC_RBRC	UC(0x005D)
#define UC_CIRM	UC(0x005E)
#define UC_UNDR	UC(0x005F)

#define UC_GRV 	UC(0x0060)

#define UC_a 	UC(0x0061)
#define UC_b 	UC(0x0062)
#define UC_c 	UC(0x0063)
#define UC_d 	UC(0x0064)
#define UC_e 	UC(0x0065)
#define UC_f 	UC(0x0066)
#define UC_g 	UC(0x0067)
#define UC_h 	UC(0x0068)
#define UC_i 	UC(0x0069)
#define UC_j 	UC(0x006A)
#define UC_k 	UC(0x006B)
#define UC_l 	UC(0x006C)
#define UC_m 	UC(0x006D)
#define UC_n 	UC(0x006E)
#define UC_o 	UC(0x006F)
#define UC_p 	UC(0x0070)
#define UC_q 	UC(0x0071)
#define UC_r 	UC(0x0072)
#define UC_s 	UC(0x0073)
#define UC_t 	UC(0x0074)
#define UC_u 	UC(0x0075)
#define UC_v 	UC(0x0076)
#define UC_w 	UC(0x0077)
#define UC_x 	UC(0x0078)
#define UC_y 	UC(0x0079)
#define UC_z 	UC(0x007A)

#define UC_LCBR	UC(0x007B)
#define UC_PIPE	UC(0x007C)
#define UC_RCBR	UC(0x007D)
#define UC_TILD	UC(0x007E)
#define UC_DEL	UC(0x007F)

#endif
`

coderkun, 2016-09-11:
@mihkell you need to [set the unicode input mode](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/keymaps/coderkun_neo2/keymap.c#L247) for your Operating System.

iFreilicht, 2016-09-12:
Try the [US International keyboard](http://dry.sailingissues.com/us-international-keyboard-layout.html) setting. It also uses AltGr to map accented versions of letters, so you only have to bind modified keys in QMK.

closed 2017-06-27 by skullydazed