qmk/qmk_firmware #747: HID Send?

IBNobody, 2016-09-12:
I am working on a way to send HID messages to AutoHotkey. ([See this post...](https://forum.pjrc.com/threads/24049-windows-dll-for-rawhid?p=115116&viewfull=1#post115116)) It is the same principle as hid_listen.

Have we implemented a way to receive data via the HID console and change keyboard parameters such as layers?

iFreilicht, 2016-09-12:
I am not aware such a mechanism even exists in the HID specification in the first place. Hid_send is implemented through reserved parts of the HID usage tables. If you find a document where sending data to an HID is documented, I'd love to know about it.

skullydazed, 2016-09-12:
We've been discussing some features that will help with that in #692.

IBNobody, 2016-09-12:
@iFreilicht 

It appears as a vendor defined usage page for PJRC. So, not part of the HID spec, but not in undefined waters.

IBNobody, 2016-09-12:
FYI: The larger usecase for my AHK system is that I want to be able to enable keyboard specific AHK hotkeys. One of the problems with AHK is that it cannot distinguish between keypresses on separate keyboards by default. There are libs that let you detect by device (AHKHID), but they do not prevent you from masking off a key on a specific keyboard. QMK and RawHID are a solution.

Here is the flow I see.

1. AHK listens to the PJRC HID channel for multiple devices.
1. I press a key on a QMK keyboard.
2. QMK recognizes that the keypress is associated with a QMK macro and fires off the associated keydown macro.
3. The QMK macro sends a character sequence via the PJRC HID channel.
4. AHK detects the character sequence coming from a specific device.
5. AHK fires off a macro tied specifically to keyboard's PJRC HID channel.

Things will get a little tricky if each keyboard has the same PID, though. AHK will know that there are two QMK devices, but it would not be able to tell them apart. Either I will have to use separate PIDs, or I will have to figure out a handshake method to get the keyboards to self-identify. Hence the need for two-way communication.

drelephant, 2016-09-13:
Well I suppose if you can get two-way communication working, you could send some kind of identifier with each message to identify each individual keyboard.

You could send a command to set an identifier in eeprom or even a randomly-generated one would be better than no identifier at all.

This doesn't answer how you'd technically achieve the communication channel, but just throwing some ideas out here.

IBNobody, 2016-09-13:
@drelephant 

Hmm... Though I have something semi-random now. Apparently the AVR USB chips we use have a serial number that is readable. I have never used it, but maybe that is something I can base things on?

By the way, the two way comm channel is already available. https://www.pjrc.com/teensy/rawhid.html

What is missing is support in QMK to do the reading (currently commented out) and then do anything with the data read.

skullydazed, 2017-06-27:
Closing this in favor of discussion on #692, #895, and a possible future qmk implementation of https://github.com/hid-io/hid-io/tree/master/spec.

closed 2017-06-27 by skullydazed