qmk/qmk_firmware #754: [kc60] mapping caps lock to LCTL

tomb0y, 2016-09-13:
Using [this](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/kc60/keymaps/workman-dead/keymap.c) keymap I can't seem to change caps lock to work as control.

For a while I thought it was working, but then it turned out it was because of the osx modifier key settings.

Am I doing something wrong in the keymap, or is this something that can't be done?

sgoodwin, 2016-09-13:
Everything else works as expected except for that one key?

tomb0y, 2016-09-13:
@sgoodwin yes, everything else works as expected, but sadly caps lock works as a caps lock, and not ctrl.

sgoodwin, 2016-09-13:
This is odd, on my kc60 I made the capslock key into an FN key with no trouble. The other control works? You're sure the board has been flashed with this change? (easy mistake to make, either not compiling again or not flashing the latest hex after the compiling)

tomb0y, 2016-09-13:
Yep, the other ctrl keys works normally.

I flashed the up to date firmware just now to be sure, but nothing changed.
I also tried setting it to `KC_A`, just to try something else out, which works, and strangely enough,  so does `KC_RCTL` which I should have tried out earlier, cause it kinda solves the problem.

I can't get `KC_LCTL` to work though.

sgoodwin, 2016-09-13:
Ah weird. Why not just use RCTL then?

tomb0y, 2016-09-13:
Yeah, that's what I'll do, thanks for being the rubber duck @sgoodwin 👍

closed 2016-09-13 by tomb0y